ros-industrial/industrial_ci #360: Replacement of `CATKIN_CONFIG`

ipa-mdl, 2019-04-20:
As part of the transition to colcon (#358) I'd like to replace `CATKIN_CONFIG` (works only with catkin-tools) by something more generic.

One use case (enabling the devel space) will get dropped, but it might make sense to support setting CMake options/variables.

Anything else?
@miguelprada, @130s, @ipa-jba

ipa-fez, 2019-04-24:
We use `CATKIN_CONFIG` to force some compiler warnings to be treated as errors:

```
CATKIN_CONFIG: '--cmake-args -DCMAKE_CXX_FLAGS="-Werror=uninitialized -Werror=return-type -Werror=format"'
```

It would be nice to still have an interface to pass arguments to CMake.

ipa-mdl, 2019-04-24:
I had the idea to  expose `CMAKE_CXX_FLAGS` and pass it to catkin and colcon.
Can you please test if `CXXFLAGS="-Werror=uninitialized -Werror=return-type -Werror=format"` (already supported) works for you as well?

ipa-fez, 2019-04-25:
We've tested it and that works for us.

miguelprada, 2019-05-09:
I've reviewed the configurations we use for our private repos and I believe we only use `CATKIN_CONFIG` for setting additional CMake args, so using `CMAKE_ARGS` would be enough for us.

I'm not sure what uses other people may have found for that variable. Reviewing the catkin docs I can only think of package blacklisting, which I'm unsure is well supported in current `master` (wouldn't it fail [here](https://github.com/ros-industrial/industrial_ci/blob/0974024ad356d7db4df07a82d4176b671df69c55/industrial_ci/src/tests/source_tests.sh#L194)?), or passing make-args.

ipa-mdl, 2019-05-15:
`CMAKE_ARGS` is implemented in #369

closed 2019-06-12 by ipa-mdl