ros-industrial/industrial_ci #375: Pass arguments to docker

kjeremy, 2019-06-06:
My gitlab instance is hardcoded in `/etc/hosts` because of reasons and the docker image cannot access it. I can work around this manually by using `docker run --network=host` which picks up the host's networking however there does not appear to be a way to pass this down using `industrial_ci`.

ipa-mdl, 2019-06-06:
Does [`DOCKER_RUN_OPTS`](https://github.com/ros-industrial/industrial_ci/blob/legacy/doc/index.rst#optional-environment-variables) not work?

kjeremy, 2019-06-06:
That doesn't seem to work unless I'm putting it in the wrong place.

```yaml
image: docker:git
services:
  - docker:dind
before_script:
  - apk add --update bash coreutils tar
  - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci
kinetic:
  script: .industrial_ci/gitlab.sh
  variables:
    ROS_DISTRO: "kinetic"
    UPSTREAM_WORKSPACE: "file"
    ROSINSTALL_FILENAME: ".rosinstall"
    DOCKER_RUN_OPTS: "--network=host"
  tags:
    - kinetic
```

Gives me:

> Running with gitlab-runner 11.11.2 (ac2a293c)
  on jkolb-multi-ros-test CZb36Udq
Using Docker executor with image docker:git ...
Starting service docker:dind ...
Pulling docker image docker:dind ...
Using docker image sha256:bed64de70fa1f4d0b5a498791647c45d954cb0306ec2852dbcfb956f4ff3b0d6 for docker:dind ...
Waiting for services to be up and running...
Pulling docker image docker:git ...
Using docker image sha256:02cac6ab7b2970f791cc69782695600f8759f14bd440be68fc371977be18b502 for docker:git ...
Running on runner-CZb36Udq-project-221-concurrent-0 via fu...
Reinitialized existing Git repository in /builds/jkolb/ros_pkg_b/.git/
Fetching changes...
fatal: unable to access 'https://gitlab-ci-token:[MASKED]@ago-git.us.ara.com/jkolb/ros_pkg_b.git/': Could not resolve host: ago-git.us.ara.com
ERROR: Job failed: exit code 1

ipa-mdl, 2019-06-06:
This fails before the invocation of `industrial_ci`..

ipa-mdl, 2019-06-06:
I guess you have to set the [network mode in your runner config](https://docs.gitlab.com/runner/configuration/advanced-configuration.html#the-runnersdocker-section)

kjeremy, 2019-06-07:
That doesn't work. Says it conflicts with "links". I'm going to close this for now since this appears to be a gitlab_ci issue.

closed 2019-06-07 by kjeremy