ros-industrial/industrial_ci #376: Have rosinstall use self-signed certificate

kjeremy, 2019-06-07:
My repo `ros_pkg_b` depends on `ros_pkg_a` which I have specified in my `.rosinstall` file:

```yaml
- git:
    local-name: pkg_a
    uri: https://ago-git.us.ara.com/jkolb/ros_pkg_a.git
    version: master
```

That git repo uses a self-signed cert and during build I get:

> >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
> Starting function 'setup_rosws'
> Writing /root/catkin_ws/src/.rosinstall
> 
> update complete.
>      Performing actions: 
> 
>      Add new elements:
>   pkg_a
> 
> Config changed, maybe you need run wstool update to update SCM entries.
> Overwriting /root/catkin_ws/src/.rosinstall
> 
> update complete.
> HOSTNAME=f87373937503
> OS_NAME=ubuntu
> TARGET_REPO_PATH=/root/src/ros_pkg_b
> TERM=dumb
> ROS_ROOT=/opt/ros/kinetic/share/ros
> OS_CODE_NAME=xenial
> ROS_PACKAGE_PATH=/opt/ros/kinetic/share
> ROS_MASTER_URI=http://localhost:11311
> ROSINSTALL_FILENAME=.rosinstall
> ROS_VERSION=1
> LC_ALL=C.UTF-8
> ICI_SRC_PATH=/root/ici
> LD_LIBRARY_PATH=/opt/ros/kinetic/lib
> CATKIN_PARALLEL_TEST_JOBS=-p4
> APTKEY_STORE_SKS=hkp://ha.pool.sks-keyservers.net
> IN_DOCKER=true
> PATH=/opt/ros/kinetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
> ROS_PARALLEL_JOBS=-j8
> UPSTREAM_WORKSPACE=file
> _DO_NOT_FOLD=true
> TARGET_REPO_NAME=ros_pkg_b
> PWD=/
> ROS_PARALLEL_TEST_JOBS=-j8
> LANG=C.UTF-8
> HASHKEY_SKS=0xB01FA116
> ROSLISP_PACKAGE_DIRECTORIES=
> DOCKER_BASE_IMAGE=ubuntu:xenial
> APTKEY_STORE_HTTPS=https://raw.githubusercontent.com/ros/rosdistro/master/ros.key
> HOME=/root
> SHLVL=1
> ROS_DISTRO=kinetic
> CATKIN_WORKSPACE=/root/catkin_ws
> PYTHONPATH=/opt/ros/kinetic/lib/python2.7/dist-packages
> PKG_CONFIG_PATH=/opt/ros/kinetic/lib/pkgconfig
> CMAKE_PREFIX_PATH=/opt/ros/kinetic
> CATKIN_PARALLEL_JOBS=-p4
> ROS_ETC_DIR=/opt/ros/kinetic/etc/ros
> _=/usr/bin/env
> [pkg_a] Fetching https://ago-git.us.ara.com/jkolb/ros_pkg_a.git (version master) to /root/catkin_ws/src/pkg_a
> Cloning into '/root/catkin_ws/src/pkg_a'...
> fatal: unable to access 'https://ago-git.us.ara.com/jkolb/ros_pkg_a.git/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
> Exception caught during install: Error processing 'pkg_a' : [pkg_a] Checkout of https://ago-git.us.ara.com/jkolb/ros_pkg_a.git version master into /root/catkin_ws/src/pkg_a failed.
> 
> ERROR in config: Error processing 'pkg_a' : [pkg_a] Checkout of https://ago-git.us.ara.com/jkolb/ros_pkg_a.git version master into /root/catkin_ws/src/pkg_a failed.
> 
> <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
> Function 'setup_rosws' returned with code '1' after 0 min 0 sec

My gitlab-runner specifies the correct crt file in it's config:

```ini
[[runners]]
  name = "jkolb-multi-ros-test"
  url = "https://ago-git.us.ara.com"
  token = "CZb36Udqn8J_zK1Jk_CS"
  tls-ca-file = "/etc/gitlab-runner/certs/ago-git.us.ara.com.crt"
  executor = "docker"
  [runners.custom_build_dir]
  [runners.docker]
    tls_verify = false
    image = "docker:git"
    privileged = true
    disable_entrypoint_overwrite = false
    oom_kill_disable = false
    disable_cache = false
    volumes = ["/cache"]
    shm_size = 0
```

but that doesn't seem to make it down.

ipa-mdl, 2019-06-07:
This is not yet implemented..
The CA key might get exposed in `CI_SERVER_TLS_CA_FILE`.
Please try setting
```
DOCKER_RUN_OPTS="-v '$CI_SERVER_TLS_CA_FILE:$CI_SERVER_TLS_CA_FILE' -e '$GIT_SSL_CAPATH=$CI_SERVER_TLS_CA_FILE'
```

Please check (with `env`) if `GIT_SSL_CAPATH` gets set by Gitlab.
If this is they case, you can even set
```
DOCKER_RUN_OPTS="-v '$GIT_SSL_CAPATH:$GIT_SSL_CAPATH' -e GIT_SSL_CAPATH
```

(There might be some issues with the quotes..)

If you find a solution, we can prepare a patch  that sets this for all Gitlab CI instances :)

kjeremy, 2019-06-07:
Using:

```yaml
image: docker:git
services:
  - docker:dind
before_script:
  - apk add --update bash coreutils tar
  - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci
kinetic:
  script:
    - env
    - .industrial_ci/gitlab.sh DOCKER_RUN_OPTS="-v '$CI_SERVER_TLS_CA_FILE:$CI_SERVER_TLS_CA_FILE' -e 'GIT_SSL_CAPATH=$CI_SERVER_TLS_CA_FILE'"
  variables:
    ROS_DISTRO: "kinetic"
    UPSTREAM_WORKSPACE: "file"
    ROSINSTALL_FILENAME: ".rosinstall"
    DOCKER_BASE_IMAGE: "ros:kinetic-ros-base" # Otherwise apt fails with a 503 fetching gcc-defaults things
    ROS_REPO: "ros"

```

GIT_SSL_CAPATH does not show up in the `env` output but `CI_SERVER_TLS_CA_FILE=/builds/jkolb/ros_pkg_b.tmp/CI_SERVER_TLS_CA_FILE`

The build fails with

> Function 'prepare_docker_image' returned with code '0' after 6 min 39 sec
> Error response from daemon: create '/builds/jkolb/ros_pkg_b.tmp/CI_SERVER_TLS_CA_FILE: "'/builds/jkolb/ros_pkg_b.tmp/CI_SERVER_TLS_CA_FILE" includes invalid characters for a local volume name, only "[a-zA-Z0-9][a-zA-Z0-9_.-]" are allowed. If you intended to pass a host directory, use absolute path
> ERROR: Job failed: exit code 1

kjeremy, 2019-06-07:
I apologize if some of this is basic: I'm pretty new to docker and gitlab ci.

kjeremy, 2019-06-07:
I think the correct syntax is

 `.industrial_ci/gitlab.sh DOCKER_RUN_OPTS="-v $CI_SERVER_TLS_CA_FILE:$CI_SERVER_TLS_CA_FILE -e GIT_SSL_CAPATH=$CI_SERVER_TLS_CA_FILE"`

which gives me `GIT_SSL_CAPATH=/builds/jkolb/ros_pkg_b.tmp/CI_SERVER_TLS_CA_FILE` before erroring out

> [pkg_a] Fetching https://ago-git.us.ara.com/jkolb/ros_pkg_a.git (version master) to /root/catkin_ws/src/pkg_a
> Cloning into '/root/catkin_ws/src/pkg_a'...
> fatal: unable to access [...]

ipa-mdl, 2019-06-07:
> I think the correct syntax is [...]

Ah, yes, sry my bad

> before erroring out

I guess it is more complicated..

ipa-mdl, 2019-06-07:
Please try
```
 DOCKER_RUN_OPTS="-v $CI_SERVER_TLS_CA_FILE:$CI_SERVER_TLS_CA_FILE -e GIT_SSL_CAPATH=$CI_SERVER_TLS_CA_FILE -v /builds/jkolb/ros_pkg_b.tmp/GIT_SSL_CAINFO:/builds/jkolb/ros_pkg_b.tmp/GIT_SSL_CAINFO -e GIT_SSL_CAINFO=/builds/jkolb/ros_pkg_b.tmp/GIT_SSL_CAINFO"
```

kjeremy, 2019-06-07:
That gives me:

> [pkg_a] Fetching https://ago-git.us.ara.com/jkolb/ros_pkg_a.git (version master) to /root/catkin_ws/src/pkg_a
> Cloning into '/root/catkin_ws/src/pkg_a'...
> fatal: unable to access 'https://ago-git.us.ara.com/jkolb/ros_pkg_a.git/': Problem with the SSL CA cert (path? access rights?)
> Exception caught during install: Error processing 'pkg_a' : [pkg_a] Checkout of https://ago-git.us.ara.com/jkolb/ros_pkg_a.git version master into /root/catkin_ws/src/pkg_a failed.
> 
> ERROR in config: Error processing 'pkg_a' : [pkg_a] Checkout of https://ago-git.us.ara.com/jkolb/ros_pkg_a.git version master into /root/catkin_ws/src/pkg_a failed.

ipa-mdl, 2019-06-07:
> Problem with the SSL CA cert (path? access rights?)

This is probably not solvable without more effort, because data must be copied.

ipa-mdl, 2019-06-07:
Another try:
```
 DOCKER_RUN_OPTS="-v $CI_SERVER_TLS_CA_FILE:$CI_SERVER_TLS_CA_FILE -e GIT_SSL_CAINFO=$CI_SERVER_TLS_CA_FILE
```

kjeremy, 2019-06-07:
Thanks. I'll try it once the apt key starts working again.

kjeremy, 2019-06-07:
Ah this is a lot closer. Now it prompts me for a username.

ipa-mdl, 2019-06-07:
> Thanks. I'll try it once the apt key starts working again.

Temporarily, you can set `HASHKEY_SKS="C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654"`

kjeremy, 2019-06-10:
I think the self signed cert issue is solved but the clone via `rosinstall` is prompting me for my password (note that I switched from `https` to `git`):

> [pkg_a] Fetching git@ago-git.us.ara.com:jkolb/ros_pkg_a.git (version master) to /root/catkin_ws/src/pkg_a
> Cloning into '/root/catkin_ws/src/pkg_a'...
> Warning: Permanently added the ECDSA host key for IP address '10.80.27.83' to the list of known hosts.
> 
> git@ago-git.us.ara.com's password: 

I have created a *global deploy key* using ED25519 according to the gitlab docs (copied the public key into gitlab) and added the deploy key to both `ros_pkg_a` and `ros_pkg_b`. I've set the `SSH_PRIVATE_KEY` variable in both repos (though I think only `ros_pkg_b` needs it?) to the corresponding private key.  I also set `SSH_SERVER_HOSTKEYS` with the output of `ssh-keyscan` like the docs say. As you can see I still get prompted for the git password.

kjeremy, 2019-06-10:
I got it to work by setting `TMPDIR`

ipa-mdl, 2019-06-11:
It might work with https as well, if you pass the token via `.netrc`: https://docs.gitlab.com/ee/user/project/new_ci_build_permissions_model.html#dependent-repositories

ipa-mdl, 2019-06-11:
Please try
```
DOCKER_RUN_OPTS="-v /etc/ssl/certs:/etc/ssl/certs:ro"
```

kjeremy, 2019-06-12:
@ipa-mdl That works!

ipa-mdl, 2019-06-12:
> That works!

So, does git ask for a username again?

kjeremy, 2019-06-12:
It does not.

ipa-mdl, 2019-06-12:
> It does not.

So, you did not test if SSL works.
Please use a `https://` URL to check.

PS: This is not needed for solving your problem, but it might help us to support self-signed certificates.

kjeremy, 2019-06-12:
Okay I misunderstood. `https://` still fails with and without that setting. `git@` actually succeeds without that settings so my original problem must have been something else.