ros-industrial/industrial_ci #377: Embed ROS gpg key?

ipa-mdl, 2019-06-07:
Adapt https://github.com/ros-infrastructure/ros_buildfarm/pull/635
Would resolve all connectivity issues.
I might implement this for #361 while porting #373

MoriKen254, 2019-06-08:
I've been facing to the following error message in CI job when using industrial_ci on both Travis CI and GitLab CI for all day today. Would you mean it could also resolve the following one?

That seems missing an authorized key from the key server.

```bash
Step 6/7 : RUN apt-key adv --keyserver "hkp://ha.pool.sks-keyservers.net" --recv-key "0xB01FA116"     || { wget "https://raw.githubusercontent.com/ros/rosdistro/master/ros.key" -O - | apt-key add -; }
 ---> Running in 70f7a1179474
Executing: /tmp/tmp.cOGE2cubZo/gpg.1.sh --keyserver
hkp://ha.pool.sks-keyservers.net
--recv-key
0xB01FA116
gpg: requesting key B01FA116 from hkp server ha.pool.sks-keyservers.net
gpg: key B01FA116: public key "Totally Legit Signing Key <mallory@example.org>" imported
gpg: key B01FA116: public key "ROS Builder <rosbuild@ros.org>" imported
gpg: Total number processed: 2
gpg:               imported: 2  (RSA: 1)
Removing intermediate container 70f7a1179474
 ---> 457f636d9bfa
Step 7/7 : RUN sed -i "/^# deb.*multiverse/ s/^# //" /etc/apt/sources.list     && apt-get update -qq     && apt-get -qq install --no-install-recommends -y         build-essential         python-catkin-tools         python-pip         python-rosdep         python-wstool         ros-kinetic-catkin         ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*
 ---> Running in 465459534b04
W: GPG error: http://packages.ros.org/ros-shadow-fixed/ubuntu xenial InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
W: The repository 'http://packages.ros.org/ros-shadow-fixed/ubuntu xenial InRelease' is not signed.
WARNING: The following packages cannot be authenticated!
  python-catkin-pkg-modules python-catkin-pkg python-osrf-pycommon
  python-catkin-tools python-rospkg-modules python-rospkg
  python-rosdistro-modules python-rosdistro python-rosdep python-vcstools
  python-wstool ros-kinetic-catkin
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
The command '/bin/sh -c sed -i "/^# deb.*multiverse/ s/^# //" /etc/apt/sources.list     && apt-get update -qq     && apt-get -qq install --no-install-recommends -y         build-essential         python-catkin-tools         python-pip         python-rosdep         python-wstool         ros-kinetic-catkin         ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'prepare_docker_image' returned with code '100' after 0 min 31 sec
ERROR: Job failed: exit code 100
```

ipa-mdl, 2019-06-08:
@MoriKen254: You are affected by #371, I will merge the fix soon. I just wanted to wait one night (timezones..)  to let the changes settle and to get some sleep ;)

MoriKen254, 2019-06-08:
@ipa-mdl 
Thanks so much! Ok, I'm enough relieved to get some sleep tonight ;)

I appreciate you guys efforts and look forward everything's gonna be back tomorrow!

Briancbn, 2020-07-07:
I will be doing this in WRID20.

PR #688 (2021-05-30): Bundle ROS repository key and use signed-by option
This bundles the new key (closes #377) and uses the future-proof way of adding the key (https://github.com/ros2/ros2_documentation/pull/1189).
closed 2021-05-30 by ipa-mdl