ros-industrial/industrial_ci #382: Retrieving artifacts from install space

kjeremy, 2019-06-10:
In my `.gitlab-ci.yml` file I try to grab the artifacts via:

```yaml
artifacts:
  paths:
    - /root/catkin_ws/install
```

The binaries do not appear to be in that directory:
> 
> cd /root/catkin_ws/build/pkg_b; catkin build --get-env pkg_b | catkin env -si  /usr/bin/make install; cd -
> 
> Finished   << pkg_b:install
> Finished  <<< pkg_b                [ 0.3 seconds ]
> [build] Summary: All 1 packages succeeded!
> [build]   Ignored:   1 packages were skipped or are blacklisted.
> [build]   Warnings:  None.
> [build]   Abandoned: None.
> [build]   Failed:    None.
> [build] Runtime: 0.3 seconds total.
> Summary: 0 tests, 0 errors, 0 failures, 0 skipped
> <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
> Function 'catkin_run_tests' returned with code '0' after 0 min 0 sec
> 
> >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
> Starting function 'catkin_install_run_tests'
> [pkg_b] Started testing...
> [pkg_b] Found 0 tests.
> <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
> Function 'catkin_install_run_tests' returned with code '0' after 0 min 0 sec
> Uploading artifacts...
> WARNING: /root/catkin_ws/install: no matching files 
> ERROR: No files to upload                          
> Job succeeded
> 
>

kjeremy, 2019-06-10:
I also tried mounting a directory via `DOCKER_RUN_OPTS=-v $HOME/install:/root/catkin_ws/install` and `mkdir` before calling `gitlab.sh` but that did not help.

ipa-mdl, 2019-06-11:
> I also tried mounting a directory via DOCKER_RUN_OPTS=-v $HOME/install:/root/catkin_ws/install

This is right way in general, but not for Docker-in-Docker.
Only few directories are shared in this case (`/build` + whatever is specified in config.toml).
(That's why `TMPDIR` needs to be changed as well..)
`DOCKER_RUN_OPTS=-v $CI_PROJECT_DIR/install:/root/catkin_ws/install` should work.

kjeremy, 2019-06-11:
That did it.

closed 2019-06-11 by kjeremy