ros-industrial/industrial_ci #383: Multiple repo workflow

kjeremy, 2019-06-11:
I have a question on workflow and setting up our Gitlab CI instance.

We currently have three repos set up.
  * `Common` (holds multiple packages pertaining to most or all robots)
  * `Robot_1` (specific packages for one robot, depends on packages in `Common`)
  * `Robot_2` (specific packages for another robot and also depends on `Common`)

What I think should happen is:
 * If `Common` is changed the following repos are built:
    * `Common`
    * `Robot_1`
    * `Robot_2`
* If `Robot_1` is changed:
    * `Robot_1`
* If `Robot_2` is changed:
    * `Robot_2`

If Robot_1 and Robot_2 are built I imagine they pull in Common via `rosinstall` files (but how do you specify multiple packages in one `rosinstall` file from one repo?) or maybe `rosdep` with a custom path? I'm not sure if the `Common` repo is built first and then artifacts are moved over into the workspace somehow?

I'm not quite sure the best way to set this up but I think it must be a somewhat common scenario. How do people support this?

miguelprada, 2019-06-11:
> If Robot_1 and Robot_2 are built I imagine they pull in Common via rosinstall files (but how do you specify multiple packages in one rosinstall file from one repo?) or maybe rosdep with a custom [...]

There's nothing that limits a .rosinstall file to declare a single repo. Quite the opposite, this is precisely designed to specify multiple source repositories with their URIs and versions, and manage them together. In this scenario, and unless `Common` is released as binary and added to the rosdistro index, `rosdep` will not help you.

Also, when running industrial_ci on a given target repository (e.g. `Robot_1`), packages pulled from a rosinstall file (e.g. `Common`) will either:
- be built together with the target packages in the same workspace (when using `legacy` branch)
- be built/installed in an _upstream_ workspace which will be _sourced_ before building the target workspace (in the upcoming version).

If you want to build and test `Robot_1` and `Robot_2` when pushing changes to `Common` you must either:
- if using `legacy` branch of industrial_ci, add `Robot_1` and `Robot_2` to a rosinstall in `Common` repo and list the packages you want to be built/tested in `PKGS_DOWNSTREAM`
- in the upcoming version, declare `Robot_1` and `Robot_2` repositories (or a rosinstall containing both) using `DOWNSTREAM_WORKSPACE`

kjeremy, 2019-06-11:
Thanks for the great info. So instead of kicking off builds for each repo involved I should instead have a `.rosinstall` file for each one that pulls in the relevant packages?

For instance a rosinstall file in `Common` that pulls in `Robot_1` and `Robot_2` and in `Robot_1` (and `Robot_2`) a rosinstall file that pulls in `Common`? That sounds reasonable.

With `PKGS_DOWNSTREAM` do I only need to specify leaf package or dependent packages as well?

Also: should I just switch to the upcoming version? (I'm assuming that's `master` and not `legacy`)

ipa-mdl, 2019-06-11:
> For instance a rosinstall file in Common that pulls in Robot_1 and Robot_2 and in Robot_1 (and Robot_2) a rosinstall file that pulls in Common? That sounds reasonable.

Exactly.

> With PKGS_DOWNSTREAM do I only need to specify leaf package or dependent packages as well?

You don't have to set this at all. In this case all other packages will get built

> Also: should I just switch to the upcoming version? (I'm assuming that's master and not legacy)

For now it is still a PR (#361), but I hope it will be merged soon :D

ipa-mdl, 2020-04-08:
answered..

closed 2020-04-08 by ipa-mdl