ros-industrial/industrial_ci #384: Service Unavailable while building images

kjeremy, 2019-06-12:
If I use the default configuration I repeatedly get "503 Service Unavailable" while downloading debs. For instance this pops up on almost every run:

> Step 7/7 : RUN sed -i "/^# deb.*multiverse/ s/^# //" /etc/apt/sources.list     && apt-get update -qq     && apt-get -qq install --no-install-recommends -y         build-essential         python-catkin-tools         python-pip         python-rosdep         python-wstool         ros-kinetic-catkin         ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*
>  ---> Running in 56a30b4e04f5
> E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/g/gcc-defaults/gcc_5.3.1-1ubuntu1_amd64.deb  503  Service Unavailable [IP: 91.189.91.23 80]
> 
> E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
> The command '/bin/sh -c sed -i "/^# deb.*multiverse/ s/^# //" /etc/apt/sources.list     && apt-get update -qq     && apt-get -qq install --no-install-recommends -y         build-essential         python-catkin-tools         python-pip         python-rosdep         python-wstool         ros-kinetic-catkin         ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
> <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
> Function 'prepare_docker_image' returned with code '100' after 2 min 37 sec
> ERROR: Job failed: exit code 100

Right now I work around this by specifying `DOCKER_BASE_IMAGE: "ros:kinetic-ros-base"` and `ROS_REPO: "ros"` in my `.gitlab-ci.yml` file but I was wondering if anyone else is seeing this and if there's something I am missing.

ipa-mdl, 2019-06-17:
This looks like a connectivity issue.. is it still not working?

kjeremy, 2019-06-17:
It still fails. Almost always with `E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/g/gcc-defaults/gcc_5.3.1-1ubuntu1_amd64.deb  503  Service Unavailable [IP: 91.189.88.162 80]`

Note that I can `wget` or `curl` that file manually just fine.

ipa-mdl, 2019-06-17:
Which config are you using?
Did you try to download this file with wget in a Dockerfile?

kjeremy, 2019-06-17:
This is my config. I can successfully download the file using `wget` from the `docker:git` image.

```
image: docker:git

# Enable docker-in-docker
services:
  - docker:dind

# The docker runner does not expose /tmp to the docker-in-docker service
# This config ensures that the temp folder is located inside the project directory (e.g. for prerelease tests or SSH agent forwarding)
# JAK: This seems to be required for the SSH agent.
variables:
  TMPDIR: "${CI_PROJECT_DIR}.tmp"
  DOCKER_DRIVER: overlay2 # Recommended when using dind for performance 

before_script:
  - apk add --update bash coreutils tar # Install industrial_ci dependencies
  - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci
kinetic:
  script:
    # Run the gitlab script.
    # Since we are using a self-signed certificate we need to mount the gitlab-runner's container's certs directory.
    # We also expose an install directory for artifacts.
    - .industrial_ci/gitlab.sh DOCKER_RUN_OPTS="-v $CI_PROJECT_DIR/install:/root/catkin_ws/install" 
  variables:
    ROS_DISTRO: "kinetic"
    UPSTREAM_WORKSPACE: "file"
    ROSINSTALL_FILENAME: ".rosinstall"
    #DOCKER_BASE_IMAGE: "ros:kinetic-ros-base" # Otherwise apt fails with a 503 fetching gcc-defaults things
    #ROS_REPO: "ros"
    GIT_SSL_NO_VERIFY: "true"
  artifacts:
    paths:
     - "$CI_PROJECT_DIR/install"
  tags:
    - kinetic
```

ipa-mdl, 2019-06-17:
I guess your DinD setup is not working properly.
Or your Docker cache needs to get  cleared

ipa-mdl, 2019-07-14:
I will close this for now. This issue is not directly related to `industrial_ci`.

closed 2019-07-14 by ipa-mdl