ros-industrial/industrial_ci #387: As an ICI develop / ICI user repo maintainer, I want the purpose of branch(es) to be clarified

130s, 2019-06-17:
# Problem
I found the default branch on GitHub is called `legacy` now. My local repo, which I hadn't updated for months (bad), tells me the default branch used to be `master`. I can't find when this happened nor the decision to do so (aplogies if there's clear notice..). Branch name and usage change, esp. when it's not announced/shared, can confuse developers and often cause unnecessary extra work.

# Possible solution
- Document the usage of branch(es).
- Since the new branching is already in effect, I don't think we need to revert the usage.

130s, 2019-06-17:
> I can't find when this happened nor the decision to do so (apologies if there's clear notice..). 

Found it. https://github.com/ros-industrial/industrial_ci/issues/358#issuecomment-496425447, I'm sorry again (apparently I'm not catching up with the development on this repo).

That said, the change needs documented at least.

ipa-mdl, 2019-06-17:
The transition to `legacy` has been [dicussed](https://github.com/ros-industrial/industrial_ci/issues/35) and [announced](https://discourse.ros.org/t/industrial-ci-migration-needed-for-recent-and-future-changes/9462).

For now, all three `branches` point to the same commit.
If you [agree](https://github.com/ros-industrial/industrial_ci/issues/358#issuecomment-501216503):
* `colcon` will get deleted
* `legacy` will point to the old version (default for some time)
* `master` will become the colcon/ROS2-ready version (#361) and eventually will become the default again

`legacy` and `master` will point to each other.

130s, 2019-06-23:
I see.

> - `legacy` will point to the old version (default for some time)

As it's [announced](https://discourse.ros.org/t/industrial-ci-migration-needed-for-recent-and-future-changes/9462) already I won't request change. But "legacy" is relative term so I'm afraid we might come up with another legacy in the future. I'd go with version number (latest ATM is 0.8.0), although it won't be a big deal at all if `legacy` branch is planned to be deleted at some point soon.

ipa-mdl, 2019-06-23:
> if legacy branch is planned to be deleted at some point soon

It should not be deleted.
It will be kept as-is without new features.
Issues will be fixed if possible with limited effort, e.g., if distros become EOL.

> But "legacy" is relative term so I'm afraid we might come up with another legacy in the future.

The name refers to legacy systems, which will not be supported by the master branch anymore.

ipa-mdl, 2019-08-17:
> That said, the change needs documented at least.

#397 will add a hint

130s, 2019-10-12:
Separately from https://github.com/ros-industrial/industrial_ci/pull/397, which I've approved, I'm afraid not all users will notice the transition even with announcement on discourse.ros.org. It might be better if industrial_ci system gives the users some hint. Maybe printing a warning message at the end when the pipeline failed etc.?

ipa-mdl, 2019-10-12:
> It might be better if industrial_ci system gives the users some hint. Maybe printing a warning message at the end when the pipeline failed etc.?

We could print a deprecation warning in all runs of `legacy` (as part of #397).
I am not sure if we should show a warning in `master`.
This will become a concern whenever we switch to the `master` version as default (End of this year?).
Instead of printing a warning for all failures, we could detect legacy options and print a warning.
Most options should be covered already, but I will go through the migration guide and check all of them.

130s, 2019-10-12:
>  I am not sure if we should show a warning in `master`.

Sorry I was talking about the branches that are/soon to-be non-default, i.e. `legacy`.

ipa-mdl, 2020-04-08:
The legacy branch is  now deprecated (#397).
Further steps are exlained in #494

closed 2020-04-08 by ipa-mdl