ros-industrial/industrial_ci #388: Error when using CMake ExternalProject_add to build third party dependency

Levi-Armstrong, 2019-06-22:
I have a pure cmake package that I am leveraging industrial_ci to build and run tests. I have a check for GTest when building tests, if GTest is not found it leverages ExternalProject_add to clone down into the workspace source directory and build the external project. It works locally but when using industrial_ci I get an error unable to change directory to /root/catkin_ws/src/googletest. 

https://travis-ci.org/Levi-Armstrong/opw_kinematics/jobs/549005026

Levi-Armstrong, 2019-06-22:
I was able to resolve the issue by cloning it into the `/tmp` directory. Not sure why it would not let me clone it into `/root/catkin_ws/src` directory.

Old
```
    ExternalProject_Add(GTest
      GIT_REPOSITORY    https://github.com/google/googletest.git
      GIT_TAG           release-1.8.1
      SOURCE_DIR        ${CMAKE_SOURCE_DIR}/../googletest
      BINARY_DIR        ${CMAKE_BINARY_DIR}/../googletest
      CMAKE_CACHE_ARGS
              -DCMAKE_INSTALL_PREFIX:STRING=${CMAKE_INSTALL_PREFIX}
              -DCMAKE_BUILD_TYPE:STRING=Release
              -DBUILD_GMOCK:BOOL=OFF
              -DBUILD_GTEST:BOOL=ON
              -DBUILD_SHARED_LIBS:BOOL=ON
    )
```

New
```
    ExternalProject_Add(GTest
      GIT_REPOSITORY    https://github.com/google/googletest.git
      GIT_TAG           release-1.8.1
      SOURCE_DIR        /tmp/googletest
      BINARY_DIR        ${CMAKE_BINARY_DIR}/../googletest
      CMAKE_CACHE_ARGS
              -DCMAKE_INSTALL_PREFIX:STRING=${CMAKE_INSTALL_PREFIX}
              -DCMAKE_BUILD_TYPE:STRING=Release
              -DBUILD_GMOCK:BOOL=OFF
              -DBUILD_GTEST:BOOL=ON
              -DBUILD_SHARED_LIBS:BOOL=ON
    )
```

ipa-mdl, 2019-06-22:
> Not sure why it would not let me clone it into /root/catkin_ws/src directory.

Because of #210 (and related..).

> I was able to resolve the issue by cloning it into the /tmp directory

It belongs into the build space.

Levi-Armstrong, 2019-06-22:
Thank you for the help. I will change it to create the source directory in the build directory.

closed 2019-06-22 by Levi-Armstrong