ros-industrial/industrial_ci #398: clang-tidy support?

gavanderhoorn, 2019-08-01:
Sort-of a follow-up to #316.

I've seen some users of `industrial_ci` manually add a clang-tidy run to their CI setup, or decide to add it to the `CMakeLists.txt` itself.

It could be nice if `industrial_ci` could do the hard work and add optional tidy support, similar to how clang-format is supported.

One big difference between tidy and format is that the latter is typically used to change things, which may not fit well with the CI theme in which `industrial_ci` fits.

gavanderhoorn, 2019-08-01:
Example of manual setup: PilzDE/pilz_industrial_motion#192 (but there are quite a few, this one is just a recent one).

Additionally, [moveit_ci](https://github.com/ros-planning/moveit_ci) also supports clang tidy (nice to see everything converging ..).

ipa-mdl, 2019-08-03:
clang-tidy can be configured like this:
```
ADDITIONAL_DEBS=clang-tidy CMAKE_ARGS="-DCMAKE_CXX_CLANG_TIDY='clang-tidy;-warnings-as-errors=*;'"
```
However, it does stop on the first errored package and it checks dependency as well.
The latter can be avoided with `TARGET_CMAKE_ARGS` in #361

> One big difference between tidy and format is that the latter is typically used to change things, which may not fit well with the CI theme in which industrial_ci fits.

This should not be a problem with #361, so we could port `moveit_ci`'s solution.

ct2034, 2019-08-08:
@ipa-mdl It stopping on the first error can be avoided by specifying a custom `.clang-tidy` file (https://github.com/PilzDE/pilz_robots/blob/melodic-devel/.clang-tidy) with an empty or no `WarningsAsErrors`. Our plan is to gradually include the specified tests into `WarningsAsErrors` as we fix them.

@gavanderhoorn Then, the benefit of our solution is that it runs for everyone withing their catkin command and future violations of the defined ruleset (which we borrowed from moveit, btw) will be directly found and error the build.

But generally, I am in favour of integrating it into industrial_ci similar to the `CLANG_FORMAT_CHECK` env

ipa-mdl, 2019-08-09:
@ct2034: Thanks for the hint, I will try it with a `.clang-tidy` as well.

ipa-mdl, 2019-08-15:
> But generally, I am in favour of integrating it into industrial_ci similar to the CLANG_FORMAT_CHECK env

I have now implemented something similar to the `CATKIN_LINT` check: #402 
It reads the settings from the build (`-DCMAKE_EXPORT_COMPILE_COMMANDS=ON`).
It is not yet bullet-proof, but it seems to work properly.

ipa-mdl, 2019-12-04:
#402 was just merged.

PR #402 (2019-12-04): implemented clang-tidy check
fixes #398 
needs #361

Can be activated with `CLANG_TIDY=true` and will only for errors.
With `CLANG_TIDY=pedantic`, it will fail for warnings as well.

The output is grouped by build spaces, which corresponds to packages except for `BUILDER=catkin_make`.

ToDo:
- [x] add proper tests
- [x] add documentation
closed 2019-12-04 by ipa-mdl