ros-industrial/industrial_ci #405: NumPy, SciPy, Pandas, etc. require Python >= 3.5

MatthijsBurgh, 2019-08-22:
The newest releases of NumPy, Scipy, Pandas, etc. require Python >= 3.5. When installing python-keras-pip with rosdep, the newest version of SciPy is installed. This fails because Python 2.7 is used

This causes our CI to fail.

I have not been able to come up with a generic solution for this.

ipa-mdl, 2019-08-22:
Were you able to install them with `rosdep` on your local system somehow?

MatthijsBurgh, 2019-08-22:
`Nope, On a clean install, I wasn't able to get it installed with just `rosdep`.

ipa-mdl, 2019-08-22:
> Nope, On a clean install, I wasn't able to get it installed with just rosdep.

So then it is an issue with rosdep.

As a work-around, you can skip the failing rosdep keys and install the dependencies with `pip3` directly.

moriarty, 2019-08-23:
https://github.com/ros/rosdistro/issues/22075

I answered on the issue above, updating pip can fix it.

ipa-mdl, 2019-08-23:
@moriarty: Thanks for pointing this out!

> updating pip can fix it.

Something like `BEFORE_SCRIPT="pip install --upgrade pip && hash -r pip"` should work then.

musamarcusso, 2019-08-23:
I had issues also with Python 2.7 packages on Travis CI. The version of `pip` available was too old, and I also used the solution `BEFORE_SCRIPT` to update it, otherwise some packages would be installed in old version or not at all. I am not entirely sure how it will work when actually releasing the package, though.

moriarty, 2019-08-23:
> I had issues also with Python 2.7 packages on Travis CI. The version of `pip` available was too old, and I also used the solution [`BEFORE_SCRIPT` to update [...]

:+1: good to know the fix is really easy, doesn't even require writing a Dockerfile!

## Off-Topic

> I am not entirely sure how it will work when actually releasing the package, though.

@musamarcusso - Just a warning, because you mention releasing... Unfortunately the ROS build farm doesn't like pip dependencies https://github.com/ros-infrastructure/ros_buildfarm/issues/643 we recently ran into the issue when we wanted to have a run-time depend on the `pip-ds4drv` (sony dual shock 4 driver) we moved the pip installation into debian post install script to ensure it's on the robots which is a work-around we can do but it's not a general solution.

https://github.com/ros-infrastructure/ros_buildfarm/issues/643#issuecomment-510934429 provided a link to a package (said it works around the issue, but I didn't try) and the ros-answer: https://answers.ros.org/question/313811/bloom-generate-debian-package-with-pip-only-dependency/

musamarcusso, 2019-08-23:
Hi @moriarty,
I had noticed that. Now I am working on a Python-only ROS package and until now I couldn't figure out a nice way to setup my dependencies either. I played around with [`catkin_virtualenv`](https://github.com/locusrobotics/catkin_virtualenv) but I ran into issues when trying to setup the dependencies for Python 3 + ROS kinetic. I made a (rather ugly) [work around](https://github.com/boschresearch/pcg_gazebo_pkgs/blob/master/pcg_libraries/_create_env.py) for the time being in the form of a script called during the build process to check for the Python packages that are missing and installing some of them in the user space, but I want to replace this for a more robust solution ASAP.

MatthijsBurgh, 2019-08-23:
> [ros/rosdistro#22075](https://github.com/ros/rosdistro/issues/22075)
> 
> I answered on the issue above, updating pip can fix it.

> I had issues also with Python 2.7 packages on Travis CI. The version of `pip` available was too old, and I also used the solution [`BEFORE_SCRIPT` to update [...]

Those two combined fixes my issue. I don't want to end your discussion. I also wouldn't say it is off-topic, but it has taken some distance to my issue. So in my opinion this issue should be closed and the discussion should continue somewhere else.

ipa-mdl, 2019-08-23:
>  So in my opinion this issue should be closed

 Agreed

closed 2019-08-23 by ipa-mdl