ros-industrial/industrial_ci #406: Failing jobs due to timeouts - Travis issue, or something else?

gavanderhoorn, 2019-09-04:
Not sure this is something that we can fix on this side, but I have quite a few jobs that have run to the max time allowed for open-source builds, such as [this one](https://travis-ci.org/ros-industrial/motoman/builds/580651197).

[This one](https://travis-ci.org/ros-industrial/motoman/builds/580723642) got very close, but didn't make it either.

[Here](https://travis-ci.org/ros-industrial/motoman/jobs/580723644), the `rosdep install` step takes 38 minutes in total.

Not sure why builds that used to take 20 minutes are suddenly taking 2.5x as long.

ipa-mdl, 2019-09-04:
Some weeks ago I saw these delays as well.
According to my tests, the apt download speed just became extremely slow.
With Travis and locally.

One workaround is to switch to a different mirror or the mirror protocol, but it turned out to be quiet unstable.

gavanderhoorn, 2019-09-05:
Let's see whether it wants to play nice again today.

I didn't really believe we could do something about `apt` being slow here, but at least wanted to track it here.

gavanderhoorn, 2019-09-05:
Perhaps Travis doesn't like me any more.

Jobs are still timing out everywhere (or not even starting), but the status pages of the service don't show anything out of the ordinary, nor are there massive nrs of tweets with complaints.

musamarcusso, 2019-09-05:
I had the exact same issue some builds ago: https://travis-ci.org/boschresearch/pcg_gazebo_pkgs/jobs/579431706

I imagine something was happening because I was trying to install a dependency `apt install libspatialindex-dev` using the `BEFORE_SCRIPT` input and after I removed it the builds came back to normal.
I wonder if this has something to do with some APT packages having to be whitelisted for Travis CI (like in the case of `libspatialindex` [here](https://github.com/travis-ci/travis-ci/issues/4560)).
By the way, I still cannot run `apt` in the `BEFORE_SCRIPT`, I just removed it for the time being so my builds don't timeout all the time.

gavanderhoorn, 2019-09-05:
@musamarcusso: thanks for the input.

Unfortunately I haven't changed anything in any of the builds that are timing out. It's just a regular CI run for a PR that adds some packages (which don't introduce new dependencies).

musamarcusso, 2019-09-05:
Interesting. I will try to add the package again and see what happens, the timeout issues lasted a couple of days and I eventually gave up.

wxmerkt, 2019-09-05:
I've seen the same for a few weeks now and I wonder if it is Travis reducing service/build times for open source projects. Our builds have doubled in time and are now running out regularly.

oscar-lima, 2019-09-05:
[We](https://travis-ci.com/KCL-Planning/rosplan_demos/builds) confirm the issue, our builds have increased in time from 17 min to 1 hour !

gavanderhoorn, 2019-09-05:
Would be nice if I can even get jobs to start :disappointed: 

But it's strange that there is no status update, nothing on their twitter, or at least not what I can see.

musamarcusso, 2019-09-05:
I also tried this `Requests` page that was introduced in their [blog post](https://blog.travis-ci.com/2014-05-12-why-is-my-build-not-running/) to see what was happening, but only the passing builds were shown, none of the ones that have this issue.

musamarcusso, 2019-09-05:
The line I see all the time when this happens is this one from `travis.sh`

```bash
echo -en "${ANSI_YELLOW}...industrial_ci is still running...${ANSI_RESET}"; }
```

ipa-mdl, 2019-09-05:
> The line I see all the time when this happens is this one from travis.sh (@musamarcusso)

Yeah, it works ;)
It is meant to circumvent the 10min silence watchdog of Travis.
Without, all jobs would probably get killed after 10 min.

> Would be nice if I can even get jobs to start disappointed (@gavanderhoorn)

I don't have any trouble to start new jobs in my fork.
The `ros-industrial` builds might queue up, because they take that long.

> I've seen the same for a few weeks now and I wonder if it is Travis reducing service/build times for open source projects (@wxmerkt)

I guess it is just some outage for some, but not all worker machines.
Our travis-ci.com jobs have a similar delay.

gavanderhoorn, 2019-09-05:
Just had a build succeed. Total time: 2 hrs +, up from around 30 mins.

oscar-lima, 2019-09-05:
We switched our CI environment from kinetic + xenial to melodic + bionic and got a reduction of half of the time (1 hour vs 30 min)

ipa-mdl, 2019-09-05:
[image]

"Experiencing Issues with Google Cloud Platform Google Compute Engine"

ipa-mdl, 2019-09-05:
> We switched our CI environment from kinetic + xenial to melodic + bionic and got a reduction of half of the time (1 hour vs 30 min)

Not all jobs are delayed. And the delays might vary a lot.
Futhermore, melodic/bionic builds are a little bit faster in generaly, because of fewer dependencies (packages got cleaned up).

gavanderhoorn, 2019-09-05:
> https://www.traviscistatus.com/: "Experiencing Issues with Google Cloud Platform Google Compute Engine"

Good to know what Travis doesn't just hate me ;)

ipa-mdl, 2019-09-06:
I ran more tests and it looks like access to `archive.ubuntu.com` is throttled for some reason.
#408 switches the mirror to EC2, but only for Travis.

ipa-mdl, 2020-04-07:
@gavanderhoorn: I guess this was resolved by Travis somehow?

oscar-lima, 2020-04-07:
is no longer an issue for us

gavanderhoorn, 2020-04-07:
Yes, I'll close this.

In the end it was likely there was some throttling *somewhere*.

But I haven't noticed it any more for a long time.

closed 2020-04-07 by gavanderhoorn