ros-industrial/industrial_ci #409: Running locally fails if docker was installed via snap?

agutenkunst, 2019-09-13:
Running industrial_ci locally failed with
```
docker command not found
```
when I had docker installed via snap. Via apt works fine now...

Can anyone reproduce?

ipa-mdl, 2019-09-13:
I did not yet use snap to install Docker.
(In the beginning it only shipped a version outdated version, )

> Running industrial_ci locally failed with

How do you run it locally? `run_ci`?
Running industrial_ci locally failed with

agutenkunst, 2019-09-13:
> How do you run it locally? `run_ci`?

run_travis

ipa-mdl, 2019-09-18:
I can confirm this issue.
In this case docker gets installed to `/snap/bin`, which is not in the default `PATH`.
Passing the user settings (#411) fixes it.

agutenkunst, 2019-10-09:
Closed by #411 .

PR #411 (2019-09-29): pass PATH in run_travis
fixes #409, should get backported to `legacy`.

@agutenkunst: please test
closed 2019-10-09 by agutenkunst