ros-industrial/industrial_ci #416: Roslint in AFTER_SCRIPT

Olavhaasie, 2019-10-09:
I am building a ROS Melodic package using colcon on Travis CI and I would like to run the roslint target inside the target repository. I have tried adding the following `AFTER_SCRIPT`:

    AFTER_SCRIPT='cd $target_ws && rosenv && colcon build --event-handlers status- desktop_notification- --cmake-target-skip-unavailable --cmake-target roslint'

based on some of the documentation available. However I get the following error when the `AFTER_SCRIPT` runs:

```
Traceback (most recent call last):
  File "/opt/ros/melodic/share/roslint/cmake/../../../lib/roslint/cpplint", line 10, in <module>
    from roslint import cpplint_wrapper
ImportError: No module named roslint
```

It seems that python cannot find the `roslint` package. It works if I run the command locally inside my workspace. I was not able to find any information on running `roslint` using `industrial_ci`, so perhaps any of you know how to implement this?

ipa-mdl, 2019-10-12:
Does it work like this in your local workspace?
Do your packages (test-)depend on `roslint`?

Olavhaasie, 2019-10-13:
The command works in my local workspace like this. The packages `<depend>` on roslint and the roslint package is installed in the `install_target_dependencies` step. It logs

    executing command [apt-get install -y -qq ros-melodic-roslint]

ipa-mdl, 2019-10-13:
Due to a bug, `rosenv && ...` did not work as expected.
Please test #421 

With the new version even `AFTER_SCRIPT='builder_run_build "$target_ws" "$target_ws" --cmake-target-skip-unavailable --cmake-target roslint'` should work.

I did not know that `roslint` requires a separate run, if `roslint_add_test()` was not added to `CMakeLists.txt`.

Olavhaasie, 2019-10-13:
I tried using `AFTER_SCRIPT='builder_run_build "$target_ws" "$target_ws" --cmake-target-skip-unavailable --cmake-target roslint'` with #421, but I still get the same error as before

ipa-mdl, 2019-10-13:
> I tried using [...], but I still get the same error as before

Sry, my bad!

I meant
```
AFTER_SCRIPT='builder_run_build "$target_ws/install" "$target_ws" --cmake-target-skip-unavailable --cmake-target roslint'
```

Your inital `AFTER_SCRIPT` works as well: https://travis-ci.org/ipa-mdl/ros_canopen/builds/597395621
(The first failes because of the bug, the others because of actual roslint errors)

Olavhaasie, 2019-10-14:
Yes that works, thank you so much! :+1:

PR #421 (2019-10-14): Fix rosenv
should fix #416 

rosenv was executed in a subshell, so `rosenv && ...` did not work.
closed 2019-10-14 by ipa-mdl