ros-industrial/industrial_ci #417: Backslash notation in .travis.yml not working with local run_travis

agutenkunst, 2019-10-10:
I needed https://github.com/PilzDE/pilz_robots/commit/0f0a2810fb813be79de6403053b4638ffec7bb8e
because when running locally via `run_travis` only the first variable would be set. And therefore e.g. there would be no AFTER_SCRIPT

`yamllint` indicates that both versions should be valid.

@ipa-mdl you opinion?

ipa-mdl, 2019-10-12:
Thanks for reporting!

I will have a look, but I doubt that we can do much about it.
Even if `yamllint` says it is correct, `PyYAML` (used by `run_travis`) and Ruby's YAML (used by Travis) don't behave the same in all cases; especially w.r.t. string handling.

ipa-mdl, 2019-10-12:
> PyYAML (used by run_travis) and Ruby's YAML (used by Travis) don't behave the same in all cases; especially w.r.t. string handling.

In this case they treat things alike: The backslash is a literal, i.e. it is a part of the string.
I could not find any hint that a backslash is needed for multi-line strings in YAML.
I guess that Travis filters this out somehow.

ipa-mdl, 2019-10-12:
> I guess that Travis filters this out somehow.

While the backslash is listed in the web view of [3016.2](https://travis-ci.org/PilzDE/pilz_robots/jobs/595933253), `travis-build` creates something like this:
```bash
travis_cmd export\ ROS_DISTRO\=\"melodic\" --echo
travis_cmd export\ CATKIN_LINT\=true --echo
travis_cmd export\ CMAKE_ARGS\=\"-DCATKIN_ENABLE_CLANG_TIDY\=true\" --echo
travis_cmd export\ ADDITIONAL_DEBS\=\"clang-tidy\ libclang-dev\" --echo
travis_cmd export\ ROS_REPO\=\"testing\" --echo
travis_cmd export\ CATKIN_LINT\=false --echo
travis_cmd export\ NOT_TEST_BUILD\=true --echo
travis_cmd export\ AFTER_SCRIPT\=\"./.coverage.sh\" --echo
travis_cmd export\ CMAKE_ARGS\=\"-DENABLE_COVERAGE_TESTING\=ON\ -DCMAKE_BUILD_TYPE\=Debug\" --echo
travis_cmd export\ DOCKER_RUN_OPTS\=\'-e\ COVERAGE_PKGS\=\"prbt_hardware_support\"\' --echo
```

I was not able to find the code for the env parser yet.
`run_ci` just passes the line as-is, without parsing the individual entries.

ipa-mdl, 2019-10-12:
> run_travis just passes the line as-is, without parsing the individual entries.

`\ AFTER_SCRIPT="./.coverage.sh"` gets parsed as `<SPACE>AFTER_SCRIPT="./.coverage.sh"` by bash.
#419 should filter `/` (and all other invalid parts).

ipa-mdl, 2020-04-07:
#419 was merged

PR #419 (2020-02-20): filter travis yaml like travis-build
fixes #417 

I am not 100% sure about the regex..
@agutenkunst: please test
closed 2020-04-07 by ipa-mdl