ros-industrial/industrial_ci #418: Release 0.9.0 or newer

130s, 2019-10-12:
0.8.0 was made in 2019/06. There's been QUITE A FEW good changes since then. Also, some features in the `master` branch are now discontinued? So it'd be good to cut a new release. Maybe incrementing major version is suitable.

Remaining items if any:
- [x] https://github.com/ros-industrial/industrial_ci/issues/333
- [ ] https://github.com/ros-industrial/industrial_ci/issues/387
- [ ] https://github.com/ros-industrial/industrial_ci/issues/358 (if we want to wait for that to be closed?)
- [ ] Else?

ipa-mdl, 2019-10-12:
I would release 0.8.1 from `legacy` (almost now new features, just some updates to reflect upstream changes), perhaps after #397.
`master` can be released as 0.9.0 right away.

New features for 0.9.1 are #415 and #402

ipa-mdl, 2020-04-07:
I will release 0.8.1 for `legacy` and 0.9.0 for `master`.

ipa-mdl, 2020-04-08:
0.9.0 and 0.10.0 are pushed

closed 2020-04-08 by ipa-mdl