ros-industrial/industrial_ci #432: Clone Up/Downstream

130s, 2019-11-02:
# Problem

For a change consists of PRs over multiple repos, each of which is in the dependency tree, a PR on target repo (say `repo_t`) that depends on the change in upstream (or often downstream as well?) to build and/or pass tests would fail unless the upstream change is available during CI process.

## When this problem may matter

For multiple repos situation described,

- Often, upstream reviewer may want to see benefit / result of the change. To show that, CI result on the downstream repo that requires the upstream change may help.
- Developer may want to use CI to run exhaustive check of her change.

# Condition of satisfaction
- [ ] Discussed and decided whether such a usecase is valid, and if a new feature is needed to achieve the usecase, it is worth maintenance.
- [ ] Implement.

130s, 2019-11-02:
> - Often, upstream reviewer may want to see benefit / result of the change. To show that, CI result on the downstream repo that requires the upstream change may help.

That's the case at Plus One so we have a very adhoc script to achieve that. It'd be great if this can be achieved via a generic feature.

What we've been doing: For the repos cloned by source via `.repos/.rosinstall`, look for the branch with the same name, from the same remote repo `repo_t` the PR is based off. E.g., say branch name used in the PR is `branch_m` on a repo `repo_a`, look for `branch_m` for all repos cloned via `.rosinstall`. If found, checkout that branch instead of the one specified in .rosinstall.

A known caveat is that the organization of up/downstream repos will be limited to the same remote repo as repo_t. Inside of corporate setting where fork is not used (like us at Plus One) it's not a problem, but on public repos this isn't the case.

130s, 2019-11-02:
Adding a case (OP to be updated):

> # When this problem may matter

- When a change in  the upstream breaks the downstream software without the change.

ipa-mdl, 2019-11-04:
Interesting idea!

Switch the branch without switching the remote is easy:

```
b=$(git -C "$TARGET_REPO_PATH" rev-parse --abbrev-ref HEAD); for p in $(find $current_ws/src -type d -name .git); do git -C $p/.. checkout $b || true; done
```
(only on master, not yet hardened, would only work on non-shallow clones)

I would be nice to have a set of hooks for common tasks like this.

> A known caveat is that the organization of up/downstream repos will be limited to the same remote repo as repo_t.

This would even work for forks, because all modification are assumed to be done in the same fork.

> That's the case at Plus One so we have a very adhoc script to achieve that

Can you share this script? It would be a good starting point.