ros-industrial/industrial_ci #434: Running CI on bare rosinstall repo

fmessmer, 2019-11-12:
I'm trying something similar as already asked in #232 - that's why same title

My repo only contains the following `travis.yml` (includes some debugging options already) and a (currently empty) `.travis.rosinstall` file (to be filled later with other repos I would like to get build and tested

```
language: generic
services:
  - docker

notifications:
  email:
    on_success: change
    on_failure: always
env:
  global:
    - ROS_REPO=ros
    - DEBUG_BASH=true
    - VERBOSE_OUTPUT=true
    - VERBOSE_TESTS=false
    - NOT_TEST_INSTALL=true
    - UPSTREAM_WORKSPACE=file
    - _GUARD_INTERVAL=10
  matrix:
    - ROS_DISTRO=kinetic
install:
  - git clone https://github.com/ros-industrial/industrial_ci.git .ci_config
script:
  - .ci_config/travis.sh
```

The output (when running `rosrun industrial_ci run_ci ROS_DISTRO=kinetic ROS_REPO=ros DEBUG_BASH=true VERBOSE_OUTPUT=true VERBOSE_TESTS=false NOT_TEST_INSTALL=true UPSTREAM_WORKSPACE=file _GUARD_INTERVAL=10`) on my repo, is the following:

```
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Starting function 'catkin_build'
++ export TARGET_PKGS
++ export PKGS_DOWNSTREAM
++ echo -e 'TARGET_PKGS: '
TARGET_PKGS: 
++ echo -e 'TARGET_REPO_PATH: /root/src/mojin_ci'
TARGET_REPO_PATH: /root/src/mojin_ci
++ '[' '' == '' ']'
+++ catkin_topological_order /root/src/mojin_ci --only-names
Workspace "/root/src/mojin_ci" seems to not contain any projects. Have you passed the correct path to a catkin workspace?++ TARGET_PKGS=
+ ici_exit
+ local exit_code=3
+ trap - EXIT
+ '[' -n catkin_build ']'
+ '[' 3 -ne 0 ']'
+ color_wrap=31
+ ici_time_end 31 3
+ '[' true ']'
+ '[' true == true ']'
+ set +x
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'catkin_build' returned with code '3' after 0 min 0 sec
+ '[' 3 == 0 ']'
+ '[' 3 == 0 ']'
+ exit 3
++ ret=3
++ trap - INT
++ '[' -n '' ']'
++ ici_quiet docker rm 617d2545ad7578f63b7fbe13baf8c11a5bb6d675e0988284aa683d52256ef4ae
++ local out
+++ mktemp
++ out=/tmp/tmp.wliZkX4gTC
++ docker rm 617d2545ad7578f63b7fbe13baf8c11a5bb6d675e0988284aa683d52256ef4ae
++ true
++ local err=0
++ '[' 0 -ne 0 ']'
++ rm -rf /tmp/tmp.wliZkX4gTC
++ return 0
++ return 3
+ ici_exit
+ local exit_code=3
+ trap - EXIT
+ '[' -n '' ']'
+ '[' 3 == 0 ']'
+ '[' 3 == 0 ']'
+ exit 3
Tests failed exit code '3'
```

Looking at #232 the `catkin build` process has already been working at some point...can't tell about the testing step for now...
However, now the script exits with `catkin_topological_order` failing - not allowing the next [if-block](https://github.com/ros-industrial/industrial_ci/blob/legacy/industrial_ci/src/tests/source_tests.sh#L203-L204) to handle this case...

Has something changed in the behavior of `catkin_topological_order` or is my `travis.yml` missing some settings?
Thanks

ipa-mdl, 2019-11-12:
https://github.com/ros-industrial/industrial_ci/pull/369 broke the support for #232 
However, this is now supported in the `master` branch.
Just set ` TARGET_WORKSPACE`, either to `file`  (deprecated)  or to the actual filename, e.g. ` TARGET_WORKSPACE=.travis.rosinstall`.

(See https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#workspace-management as well)

ipa-mdl, 2019-11-12:
>     - _GUARD_INTERVAL=10

Don't do this, it it will pollute your logs (10 seconds..).
Just stick with the default (9 minutes).

fmessmer, 2019-11-12:
> #369 broke the support for #232
> However, this is now supported in the `master` branch.
> Just set ` TARGET_WORKSPACE`, either to `file` (deprecated) or to the actual filename, e.g. ` [...]

just to get you right:
 - my usecase cannot be configured on the `legacy` branch
 - on `master` branch, I have to add `TARGET_WORKSPACE=.travis.rosinstall` to the respective travis job
correct?

ipa-mdl, 2019-11-12:
> correct

Yes. Did it work?

fmessmer, 2019-11-12:
> > correct
> 
> Yes. Did it work?

Thanks for the confirmation!
I'm still playing with it...in between slots of other stuff...will report back here asap

fmessmer, 2019-11-12:
while it seems to work with a minimal `.travis.rosinstall` (single, simple, public repo using `https` entry in the `.travis.rosinstall`)...it does not work with a similar, single, simple - but private - repo using `ssh` entry in the `.travis.rosinstall`...
has something changed dealing with private repos in the `.travis.rosinstall`?

```
- git:
    local-name: mojin_supported_robots
    uri: git@github.com:mojin-robotics/mojin_supported_robots
    version: indigo-devel
```
```
=== /root/target_ws/src/mojin_supported_robots (git) ===
Could not clone repository 'git@github.com:mojin-robotics/mojin_supported_robots': Cloning into '.'...
error: cannot run ssh: No such file or directory
fatal: unable to fork
+ ici_exit
+ local exit_code=1
+ trap - EXIT
+ '[' -n setup_target_workspace ']'
+ '[' 1 -ne 0 ']'
+ color_wrap=31
+ ici_time_end 31 1
+ '[' true ']'
+ '[' true == true ']'
+ set +x
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'setup_target_workspace' returned with code '1' after 0 min 17 sec
```
```
- git:
    local-name: cob_supported_robots
    uri: https://github.com/ipa320/cob_supported_robots
    version: indigo_dev
```
```
...
+ exit 0
All tests succeeded
```

ipa-mdl, 2019-11-12:
> error: cannot run ssh: No such file or directory

This might be a problem with the ` master` branch in general.
Does `ADDITIONAL_DEBS=openssh-client` fix the build?
We should install this automatically later, whenver ` .ssh`  is present in the container..

fmessmer, 2019-11-12:
yes, `ADDITIONAL_DEBS=openssh-client` helps for the minimal example...testing our full usecase...
what needs to be done to get ssh enabled without `ADDITIONAL_DEBS=openssh-client`?

ipa-mdl, 2019-11-12:
> what needs to be done to get ssh enabled without ADDITIONAL_DEBS=openssh-client?

https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/workspace.sh needs to be extended to install `ssh-client` if it is needed, but not present.

fmessmer, 2019-11-13:
while #435 fixes the `ssh` issue, I'm now facing that `pip` is missing:
```
Starting function 'install_target_dependencies'
/usr/bin/python: No module named pip
debconf: delaying package configuration, since apt-utils is not installed
/usr/bin/python: No module named pip
ERROR: the following rosdeps failed to install
  pip: command [/usr/bin/python -m pip install -U -q pypng] failed
```

should I fix this via `ADDITIONAL_DEPS` or do you want to add this to `industrial_ci` somewhere as well?

ipa-mdl, 2019-11-13:
>  I'm now facing that pip is missing

Oh, it is missing from the official ROS docker images..
I have updated #435

fmessmer, 2019-11-18:
Summary:
 - using `industrial_ci/master`:
   1. simply use `TARGET_WORKSPACE=.travis.rosinstall`
 - using `industrial_ci/legacy`:
   1. not fail catkin_topological_order for empty workspace (see https://github.com/ros-industrial/industrial_ci/pull/436)
   2. add a `meta-package` to your "bare" repo declaring `exec_depend` on your "downstream-most" packages of the `.travis.rosinstall` workspace

ipa-mdl, 2019-12-04:
Anything else to improve or discuss? If not, please close this issue :)

fmessmer, 2019-12-04:
closing this issue as possible solutions are documented

closed 2019-12-04 by fmessmer