ros-industrial/industrial_ci #444: [master] weird run_target_tests failure

fmessmer, 2019-12-06:
I get a really weird behavior when using `industrial_ci/master` branch on one of our repos:

I have a `.rosinstall` file and I'm testing it against `ROS_DISTRO=kinetic` and `ROS_DISTRO=melodic`.
The same `.rosinstall` job succeeds for `ROS_DISTRO=kinetic` - but fails for `ROS_DISTRO=melodic` **after** the `run_target_tests` step....
The same `.rosinstall` job succeeds when running it locally on my laptop and also when using the same `.rosinstall` file with a job using `industrial_ci/legacy` branch - again for both `ROS_DISTRO=kinetic` and `ROS_DISTRO=melodic`.

I absolutely cannot see what's wrong.
Here is an excerpt of the log output:
```
...

[build] Summary: All 174 packages succeeded!

[build]   Ignored:   9 packages were skipped or are blacklisted.

[build]   Warnings:  31 packages succeeded with warnings.

[build]   Abandoned: None.

[build]   Failed:    None.

[build] Runtime: 41 minutes and 59.6 seconds total.

<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

Function 'run_target_test' returned with code '0' after 42 min 0 sec

Full test results for 'build/msh_statemachine/test_results/msh_statemachine/rostest-unittest_state_machine__robot_cob4-14__robot_env_mm-zuerich-sihlcity.xml'

-------------------------------------------------

<?xml version="1.0" encoding="utf-8"?>

<testsuite errors="0" failures="1" name="unittest.suite.TestSuite" tests="1" time="2.834"><testcase classname="rostest.runner.RosTest" name="testunittest_state_machine_mm-zuerich-sihlcity" time="2.8333"><failure type="AssertionError">test [unittest_state_machine_mm-zuerich-sihlcity] did not generate test results

  File "/usr/lib/python2.7/unittest/case.py", line 329, in run

    testMethod()

  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostest/runner.py", line 164, in fn

    self.assert_(os.path.isfile(test_file), "test [%s] did not generate test results"%test_name)

  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue

    raise self.failureException(msg)

</failure></testcase><system-out>&lt;![CDATA[

[ROSTEST]setup[/root/target_ws/src/msh/msh_statemachine/test/unittest_state_machine.test] run_id[09e6e3e6-1797-11ea-9e17-0242ac110002] starting

&amp;#x001B;[1mstarted roslaunch server http://20e79fac3b0a:44209/&amp;#x001B;[0m

SUMMARY

========

PARAMETERS

 * ...

NODES

[ROSTEST]setup[/root/target_ws/src/msh/msh_statemachine/test/unittest_state_machine.test] run_id[09e6e3e6-1797-11ea-9e17-0242ac110002] done

[ROSTEST]Running test [unittest_state_machine_mm-zuerich-sihlcity]

&amp;#x001B;[1mROS_MASTER_URI=http://20e79fac3b0a:45837/&amp;#x001B;[0m

&amp;#x001B;[1msetting /run_id to 09e6e3e6-1797-11ea-9e17-0242ac110002&amp;#x001B;[0m

[ROSTEST]removing previous test results file [/root/target_ws/build/msh_statemachine/test_results/msh_statemachine/rosunit-unittest_state_machine_mm-zuerich-sihlcity.xml]

[ROSTEST]running test unittest_state_machine_mm-zuerich-sihlcity

&amp;#x001B;[1mprocess[unittest_state_machine_mm-zuerich-sihlcity-1]: started with pid [30407]&amp;#x001B;[0m

&amp;#x001B;[1m[unittest_state_machine_mm-zuerich-sihlcity-1] process has finished cleanly

log file: /root/.ros/log/09e6e3e6-1797-11ea-9e17-0242ac110002/unittest_state_machine_mm-zuerich-sihlcity-1*.log&amp;#x001B;[0m

[ROSTEST]test [unittest_state_machine_mm-zuerich-sihlcity] finished

[ROSTEST]tearDown[/root/target_ws/src/msh/msh_statemachine/test/unittest_state_machine.test]

shutting down processing monitor...

... shutting down processing monitor complete

[ROSTEST]rostest teardown /root/target_ws/src/msh/msh_statemachine/test/unittest_state_machine.test complete

]]&gt;</system-out><system-err>&lt;![CDATA[

&amp;#x001B;[31mWARNING: WARN: illegal &lt;node&gt; name 'unittest_state_machine_mm-zuerich-sihlcity'.

http://ros.org/wiki/Names

This will likely cause problems with other ROS tools.

Node xml is &lt;test pkg="msh_statemachine" test-name="unittest_state_machine_$(arg robot_env)" type="unittest_state_machine.py"&gt;

        &lt;rosparam command="load" file="$(find msh_config)/envs/$(arg robot_env)/sm_settings.yaml" ns="settings"/&gt;

        &lt;rosparam command="load" file="$(find msh_config)/envs/$(arg robot_env)/statemachine/tts_sentences.yaml" ns="tts_sentences"/&gt;

        &lt;rosparam command="load" file="$(find msh_config)/envs/$(arg robot_env)/statemachine/tts_monologs.yaml" ns="tts_monologs"/&gt;

        &lt;rosparam command="load" file="$(find msh_config)/envs/$(arg robot_env)/statemachine/offers.yaml" ns="offers"/&gt;

        &lt;rosparam command="load" file="$(find msh_config)/envs/$(arg robot_env)/statemachine/quiz.yaml" ns="quiz"/&gt;

        &lt;rosparam command="load" file="$(find msh_config)/envs/$(arg robot_env)/statemachine/poi.yaml" ns="pois"/&gt;

    &lt;/test&gt;&amp;#x001B;[0m

]]&gt;</system-err></testsuite>

-------------------------------------------------

build/msh_statemachine/test_results/msh_statemachine/rostest-unittest_state_machine__robot_cob4-14__robot_env_mm-zuerich-sihlcity.xml: 1 tests, 0 errors, 1 failures, 0 skipped

Summary: 636 tests, 0 errors, 1 failures, 0 skipped

The command ".industrial_ci/travis.sh" exited with 1.

Done. Your build exited with 1.
```

The "Warning" about illegal node name is known...and not causing the test to fail!

What makes it even weirder is the fact that it's not always the same `rostest_unittest_X.xml` that fails...we are testing various `robot` and `robot_env` combos...and in case a `robot` is not configured correctly, there would be several `rostest_unittest_X.xml` failing - same for `robot_env`.

It also seems to me that the `run_target_tests` step already finished successfully.

Thus:
Does anybody spot a reason for the test failing?
What's happening after `run_target_tests` that might be causing the failure?

Hoping to get some help here

ipa-mdl, 2019-12-06:
> after the run_target_test step....
> the failure seems to be reported by catkin_test_results --verbose

Building / running the tests never fails (exit code !=0) for test failures.
Only `catkin_test_result` reports a non-zero exit code.
This is it how it was designed.

It fails because of
```
    self.assert_(os.path.isfile(test_file), "test [%s] did not generate test results"%test_name)
```

> The same .rosinstall job succeeds when running it locally on my laptop

locally tested with `run_ci` / `run_travis`?

fmessmer, 2019-12-06:
locally tested with `rosrun industrial_ci run_ci`...

fmessmer, 2019-12-08:
I might have to update my report:
I am able to get my jobs (both `kinetic` and `melodic`) to succeed by restarting them a couple of times...

Thus, the test failures seem kind of random.
But it's always a `unittest` like `add_rostest(test/unittest_state_machine.test ARGS robot:=${robot} robot_env:=${env})` that fails - it's never a `roslaunch_add_file_check`...
I don't have any issues with these tests on the respective build using `industrial_ci/legacy` branch

ipa-mdl, 2019-12-09:
> Thus, the test failures seem kind of random.

It looks like a resource issue..
Do you configure job control?

fmessmer, 2019-12-09:
what do you mean? `X_PARALLEL_JOBS`?
I thought this is no longer supported un `master`
or is there a travis-native setting?

ipa-mdl, 2019-12-09:
> what do you mean? X_PARALLEL_JOBS?
I thought this is no longer supported un master
or is there a travis-native setting?

That's why I am asking ;)
As far as I can tell his is the only part that changed for ROS1.

If a test did not produce test results, it might have crashed, e.g. because no memory is left..

fmessmer, 2019-12-09:
yes, I had `X_PARALLEL_JOBS=1` configured for `industrial_ci/legacy` builds - but only for `melodic`...and the reason was a package could not allocate enough memory during `catkin build` phase - never had a memory issue with `kinetic`...never had an issue with the testing phase...but for `melodic` that might have been hidden due to the already set `X_PARALLEL_JOBS=1`

Is there a way to still set the parallel jobs? (for testing phase only?)

ipa-mdl, 2019-12-09:
> Is there a way to still set the parallel jobs? (for testing phase only?)

In code ;)
Somewhere around here:
https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/builders/catkin_tools.sh#L43

ipa-mdl, 2019-12-09:
> never had a memory issue with kinetic...never had an issue with the testing phase

The `legacy`  sets up job control with default values, but they should be higher than the defaults of `catkin-tools`.

There should be some log output for the crashed node..

fmessmer, 2019-12-19:
fixed via #446

PR #446 (2019-12-19): introduce PARALLEL_TESTS for builder_run_tests (all builders)
referring to issue https://github.com/ros-industrial/industrial_ci/issues/444
fixes #444 

as the assumption for the failures during `builder_run_tests` is "memory issue", I'm trying to limit the concurrent jobs by setting `--cmake-args -j1` for the `builder_run_tests` phase

for this, I introduced a new setting `TEST_CMAKE_ARGS` (to be distinguished from `*CMAKE_ARGS` for the `builder_run_build` phase)

by setting `TEST_CMAKE_ARGS=-j1`, I expect the command to be extended to 
```
catkin build --catkin-make-args run_tests -- --no-st [... cut]
closed 2019-12-19 by fmessmer