ros-industrial/industrial_ci #445: gitlab example code fails on gitlab

tylerjw, 2019-12-11:
I tried using the example `.gitlab-ci.yml` from the README.md and it is failing due to `apk` missing:

```bash
bash: line 90: apk: command not found
```

Here is my config (copied from README.md):

```yml
image: docker:git
services:
  - docker:dind
before_script:
  - apk add --update bash coreutils tar
  - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci
melodic:
  script: .industrial_ci/gitlab.sh ROS_DISTRO=melodic
```

tylerjw, 2019-12-12:
This is because it is running as the Shell executor.  I just need to figure out how to get it to run with the docker executor.  Leaving this here in case others have this issue.

ipa-mdl, 2019-12-12:
> This is because it is running as the Shell executor.

In principle, it should work with the shell executor as well (if you have Docker installed).
It might just work by removing that `apk` line.

tylerjw, 2019-12-12:
@ipa-mdl It doesn't work by just removing the apk line.  It fails with this error:

```bash
SSH auto set-up cannot be used in non-disposable environments
```

ipa-mdl, 2019-12-12:
> SSH auto set-up cannot be used in non-disposable environments

This gets printed because you have specified `SSH_PRIVATE_KEY`, which is only needed (and therefore supported) for the docker executor.
If you really need to setup SSH keys, try this: https://docs.gitlab.com/ee/ci/ssh_keys/#ssh-keys-when-using-the-shell-executor

tylerjw, 2019-12-12:
That did work.  I first took the tactic of trying to get a docker executor and then the system person managing the infrastructure objected because that executor requires escalated privileges to run.  How have you navigated the security concerns of the docker executor?

tylerjw, 2019-12-12:
Also, I'm now getting a failure about missing packages that should be gotten by wstool from the rosinstall.  I updated the variable `ROSINSTALL_FILENAME` but it is still failing.

ipa-mdl, 2019-12-13:
> I updated the variable ROSINSTALL_FILENAME but it is still failing.

Did you set `UPSTREAM_WORKSPACE=file` (literal!) as well?
In general, I would recommend to use the `master` branch, in which you can set the filename directly: `UPSTREAM_WORKSPACE=my_repo.rosinstall`

tylerjw, 2019-12-13:
I'm confused by what you are suggesting.  I tried setting that variable and read the code around how wstool is used and am confused why it isn't working.  Here is my `.gitlab-ci.yml`:

```yml
image: docker:git
services:
  - docker:dind
before_script:
  - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci
melodic:
  script:
    - export UPSTREAM_WORKSPACE=my_project.rosinstall
    - export ROSINSTALL_FILENAME=my_project.rosinstall
    - .industrial_ci/gitlab.sh ROS_DISTRO=melodic
```

tylerjw, 2019-12-13:
I didn't realize I wasn't checking out the master branch.  It seems like there should be a warning at the top of the readme if the default branch is not the master branch.  Something like:

**<span style="color:red">WARNING</span>: The default branch is not the master branch, it is a legacy branch and all new users are encouraged to use the latest code in the master branch**

ipa-mdl, 2019-12-13:
> It seems like there should be a warning at the top of the readme if the default branch is not the master branch

Yes, this is pending: #397
However, both versions work perfectly fine with Gitlab.

tylerjw, 2019-12-13:
My latest problem is I am exceeding the job log limit (project depends on feature branch of moveit not yet merged so it has to build moveit from source).  I noticed one thing really filling up the logs is build status prints.  Shouldn't catkin be run with `--no-status` my default ?

tylerjw, 2019-12-13:
It looks like it is using `catkin_make` from reading the code :man_facepalming: :(

ipa-mdl, 2019-12-13:
> It looks like it is using catkin_make from reading the code man_facepalming :(

[Better read again..](https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/env.sh#L65)  `catkin_make` support is a recent addition, and not the default.

> I found the no-status flag used on some random commit hash that is not part of a branch.

`catkin-tools` is [run with `--no-status`](https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/builders/catkin_tools.sh#L30), but as well with `-vi`. Please set `VERBOSE_OUTPUT=false` to suppress this.
Right now I am not sure, why I changed the default to verbose output. I will have a look again.

Please note that the `master` branch is a heavily refactored version, so minor glitches are to be expected. It is not yet as battle-tested as the `legacy` branch (which is therefore still the default).

tylerjw, 2019-12-13:
I went back to moveit_ci and it just worked, we merged it, and I'm now working on something else.  I might revisit this in the future.  Your system here has many features but it seems that using the default options are broken and/or poorly documented.  Also, the one feature I was really hoping for (gitlab log folding) doesn't seem to work anyway (unless there is yet another environment variable I need to set to enable this).

ipa-mdl, 2019-12-13:
> I went back to moveit_ci and it just worked, we merged it,

Usually, `industrial_ci` "just works" as well ;)
But every project is a little bit different, so our defaults might not work in all cases.
If `moveit_ci` (which as well adapted to MoveIt!) works for you, then it is fine,

> Your system here has many features but it seems that using the default options are broken and/or poorly documented.

Our defaults are not broken, and all variables are documented.
However, for sure we need better getting-started guides. Contributions are more than welcome!

> Right now I am not sure, why I changed the default to verbose output. I will have a look again.

If I am not mistaken, `colcon` defaults to verbose mode and I did not find a way yet to suppress it properly. This change is actually not documented yet, this just slipped through.

ipa-mdl, 2019-12-13:
Since there is no actual problem with Gitlab shell executors and `industrial_ci`, I will close this issue now,

closed 2019-12-13 by ipa-mdl