ros-industrial/industrial_ci #450: How to run catkin test as part of CI?

Levi-Armstrong, 2019-12-14:
I tried using the AFTER_SCRIPT="catkin test --no-deps package_name" but I receive an error unable to find source space. The exact error print out is below.

```
Starting function 'after_script'

==> Expanding alias 'test' from 'catkin test --no-deps tesseract_collision tesseract_environment tesseract_geometry tesseract_kinematics tesseract_motion_planners tesseract_process_planners tesseract_scene_graph tesseract_urdf' to 'catkin build --verbose --make-args test -- --no-deps tesseract_collision tesseract_environment tesseract_geometry tesseract_kinematics tesseract_motion_planners tesseract_process_planners tesseract_scene_graph tesseract_urdf'

[build] Error: Unable to find source space `/root/src/tesseract/src`
```

ipa-mdl, 2019-12-14:
All scripts are run in the source folder, just as with Travis.
For catkin you have to pass the workspace path as well, e.g.
```
AFTER_SCRIPT="catkin test -w ~/catkin_ws --no-deps package_name"
```
(legacy version, master uses different workspaces)

Levi-Armstrong, 2019-12-14:
@ipa-mdl Thank you! That fixed it.

closed 2019-12-14 by Levi-Armstrong