ros-industrial/industrial_ci #455: apt-key adv behind corporate proxy

kledom, 2019-12-19:
Hello,

I'm trying to run industrial_ci behind a corporate proxy. After reading [https://docs.gitlab.com/runner/configuration/proxy.html](url) i was able to solve the usual pain that is always involved when working behind proxy servers. However, there is one problem I couldn't fix so far: I think that there is currently (Ubuntu 18.04) a bug in apt-key adv [https://bugs.launchpad.net/ubuntu/+source/gnupg2/+bug/1788190](url), which is why I get a timout on the command setting up the ROS keys.
Is this a known problem and does anybody know a workaround?

ipa-mdl, 2019-12-19:
> Is this a known problem and does anybody know a workaround?

I don't know this problem (no corporate proxy...), but you could try setting

`APTKEY_STORE_HTTPS=http://packages.ros.org/ros.key`. This will download the key with `wget` instead of `apt-key adv`.
(Beware: not served via HTTPS, only HTTP)

Or we could implement #377, then this step would not be needed at all.

kledom, 2019-12-20:
> `APTKEY_STORE_HTTPS=http://packages.ros.org/ros.key`. This will download the key with `wget` instead of `apt-key adv`.
> (Beware: not served via HTTPS, only HTTP)

Thank you! This seems to have fixed the problem regarding the key. 

However, I'm still getting a error in the command
`RUN sed -i "/^# deb.*multiverse/ s/^# //" /etc/apt/sources.list     && apt-get update -qq     && apt-get -qq install --no-install-recommends -y         build-essential         python-catkin-tools         python-pip         python-rosdep         python-wstool         ros-melodic-catkin         ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*`

```
 ---> Running in 05c77576fc56
--2019-12-20 09:59:58--  http://packages.ros.org/ros.key
Resolving <proxy> (<proxy>)... 153.96.12.17
Connecting to <proxy> (<proxy>)|153.96.12.17|:3128... connected.
Proxy request sent, awaiting response... 200 OK
Length: 1167 (1.1K)
Saving to: 'STDOUT'

     0K .                                                     100%  266M=0s

2019-12-20 09:59:58 (266 MB/s) - written to stdout [1167/1167]

Warning: apt-key output should not be parsed (stdout is not a terminal)
OK
Removing intermediate container 05c77576fc56
 ---> 3c909badda3a
Step 7/7 : RUN sed -i "/^# deb.*multiverse/ s/^# //" /etc/apt/sources.list     && apt-get update -qq     && apt-get -qq install --no-install-recommends -y         build-essential         python-catkin-tools         python-pip         python-rosdep         python-wstool         ros-melodic-catkin         ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*
 ---> Running in c9b554c26312
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/bionic/InRelease  Connection failed [IP: 153.96.12.17 3128]
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/bionic-updates/InRelease  Connection failed [IP: 153.96.12.17 3128]
W: Failed to fetch http://security.ubuntu.com/ubuntu/dists/bionic-security/InRelease  Connection failed [IP: 153.96.12.17 3128]
W: Failed to fetch http://packages.ros.org/ros-testing/ubuntu/dists/bionic/InRelease  Connection failed [IP: 153.96.12.17 3128]
W: Some index files failed to download. They have been ignored, or old ones used instead.
E: Unable to locate package python-catkin-tools
E: Unable to locate package ros-melodic-catkin
The command '/bin/sh -c sed -i "/^# deb.*multiverse/ s/^# //" /etc/apt/sources.list     && apt-get update -qq     && apt-get -qq install --no-install-recommends -y         build-essential         python-catkin-tools         python-pip         python-rosdep         python-wstool         ros-melodic-catkin         ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
```

ipa-mdl, 2019-12-20:
Did you try https://github.com/ros-industrial/industrial_ci/pull/310#issuecomment-409195307?

closed 2020-12-13 by ipa-mdl