ros-industrial/industrial_ci #459: Skip launching docker image

tylerjw, 2019-12-29:
Because of security concerns around dind CI builds in gitlab I would like to let gitlab handle starting the image and just run CI inside of it.  This might already exist and I just haven't figured out how to do it.

ipa-mdl, 2020-01-24:
> Reading the code I found the test around the variable IN_DOCKER

Please be aware that setting `IN_DOCKER=false` bypasses some code, e.g. for setting up the ROS repository.

I have plans to refactor this part of the code.
Eventually, `IN_DOCKER` should get replaced by something else as well.
I could even imagine that we introduce an option to choose the "executor".

>  If my usecase is normal enough maybe I could make a small PR to the documentation to document setting it to skip launching docker?

`IN_DOCKER` is not documented, because it could lead to unintended side effects, especially if someone would use it locally or even worse with - for some reason - password-less sudo.

I would prefer a more fail-safe option: some special handing in `gitlab.sh`.
For example a flag, which would test if the scripts is run in a disposable environment and would then set `IN_DOCKER=true`.

tylerjw, 2020-06-06:
One more comment on this is that I've found it useful to use `IN_DOCKER` locally to allow me to manage the docker launch and some environment issues I haven't found a way around otherwise.  For context here is my notes on doing that:

```bash
# pull and open docker image
docker pull moveit/moveit:melodic-ci
docker run -it -v $HOME/.ssh:/root/.ssh-keys:ro -v $HOME/.ccache:/root/.ccache --cap-add=SYS_PTRACE --security-opt seccomp=unconfined moveit/moveit:melodic-ci /bin/bash

# configure .ssh folder
mkdir -p ~/.ssh
cp -r /root/.ssh-keys/* ~/.ssh/
chown -R $(id -u):$(id -g) ~/.ssh

# Update and Setup enviroment
apt update -qq
apt dist-upgrade -qq --yes
apt install -qq --yes openssh-client apt-utils gnupg wget ca-certificates lsb-release dirmngr vim
echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections
eval $(ssh-agent -s) && ssh-add ~/.ssh/id_rsa

# Setup test workspace
CATKIN_WS=$HOME/tmp/ws_test/
mkdir -p $CATKIN_WS/src
cd $CATKIN_WS/src
git clone https://github.com/ros-industrial/industrial_ci.git -b master
git clone ssh://git@gitlab.company.com:8082/org/project_repo.git
cd $CATKIN_WS
catkin config --install
catkin build industrial_ci
source install/setup.bash
cd src/project_repo

# Install dependencies
./install_dependencies.sh
source /opt/intel/openvino/bin/setupvars.sh

# run ci
rosrun industrial_ci run_ci ROS_DISTRO=melodic ROS_REPO=ros DOCKER_IMAGE=moveit/moveit:master-ci \
    ROS_LANG_DISABLE=genlisp CMAKE_ARGS="-DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS_RELEASE=-O3" \
    UPSTREAM_WORKSPACE='upstream.rosinstall' TARGET_WORKSPACE='. project_repo.rosinstall' \
    CCACHE_DIR="$HOME/.ccache" IN_DOCKER=1
```

tylerjw, 2020-06-11:
My use for this was mostly fixed by the feature to enable sourcing the openvino stuff.  The only reason I'm continuing to depend on this is a customer's gitlab setup.  Out of security concerns, they are only supporting docker runners without dind support.

ipa-mdl, 2020-12-13:
#558 is the first step to enable this

ipa-mdl, 2021-01-27:
This has been implemented in https://github.com/ros-industrial/industrial_ci/pull/572

closed 2021-01-27 by ipa-mdl