ros-industrial/industrial_ci #465: run_travis fails during setup_upstream_workspace with private repos while run_ci does not

fmessmer, 2020-01-18:
testing locally, `run_travis` fails to setup_upstream_workspace with the following rosinstall file containing a private repo
```
- git:
    local-name: mojin_supported_robots
    uri: git@github.com:mojin-robotics/mojin_supported_robots.git
    version: indigo-devel
```
while `run_ci` - with the same job options works
is it only me?
any idea?

```
( | )fxm@alpamayo:~/projects/default/robot_ws/src/mojin_calibration_data (rosenv_after_script)$ rosrun industrial_ci run_travis 1
Running job 1: /home/fxm/projects/ci/install/lib/industrial_ci/run_ci /home/fxm/projects/default/robot_ws/src/mojin_calibration_data ADDITIONAL_DEBS='apt-utils dialog curl wget' AFTER_SCRIPT='rosenv sh .travis.xacro_test.sh' CATKIN_LINT=pedantic CMAKE_ARGS=-DCMAKE_BUILD_TYPE=Release NOT_TEST_DOWNSTREAM=true ROS_REPO=ros UPSTREAM_WORKSPACE=file VERBOSE_OUTPUT=false VERBOSE_TESTS=false ROS_DISTRO=kinetic
Testing /home/fxm/projects/default/robot_ws/src/mojin_calibration_data
Replacing 'UPSTREAM_WORKSPACE=file' with 'UPSTREAM_WORKSPACE=.travis.rosinstall'

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Starting function 'prepare_docker_image'
Pulling Docker image 'ros:kinetic-ros-core'...
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'prepare_docker_image' returned with code '0' after 0 min 3 sec

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Starting function 'init'
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package libapt-inst2.0:amd64.
(Reading database ... 42390 files and directories currently installed.)
Preparing to unpack .../libapt-inst2.0_1.2.32_amd64.deb ...

Unpacking libapt-inst2.0:amd64 (1.2.32) ...
Selecting previously unselected package apt-utils.
Preparing to unpack .../apt-utils_1.2.32_amd64.deb ...
Unpacking apt-utils (1.2.32) ...
Selecting previously unselected package wget.
Preparing to unpack .../wget_1.17.1-1ubuntu1.5_amd64.deb ...
Unpacking wget (1.17.1-1ubuntu1.5) ...
Selecting previously unselected package curl.
Preparing to unpack .../curl_7.47.0-1ubuntu2.14_amd64.deb ...
Unpacking curl (7.47.0-1ubuntu2.14) ...
Selecting previously unselected package dialog.
Preparing to unpack .../dialog_1.3-20160209-1_amd64.deb ...
Unpacking dialog (1.3-20160209-1) ...
Processing triggers for libc-bin (2.23-0ubuntu11) ...
Setting up libapt-inst2.0:amd64 (1.2.32) ...
Setting up apt-utils (1.2.32) ...
Setting up wget (1.17.1-1ubuntu1.5) ...
Setting up curl (7.47.0-1ubuntu2.14) ...
Setting up dialog (1.3-20160209-1) ...
Processing triggers for libc-bin (2.23-0ubuntu11) ...
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'init' returned with code '0' after 0 min 32 sec

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Starting function 'catkin_tools_setup'
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'catkin_tools_setup' returned with code '0' after 0 min 9 sec

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Starting function 'setup_rosdep'
Selecting previously unselected package python-pip-whl.
(Reading database ... 42911 files and directories currently installed.)
Preparing to unpack .../python-pip-whl_8.1.1-2ubuntu0.4_all.deb ...
Unpacking python-pip-whl (8.1.1-2ubuntu0.4) ...
Selecting previously unselected package python-pip.
Preparing to unpack .../python-pip_8.1.1-2ubuntu0.4_all.deb ...
Unpacking python-pip (8.1.1-2ubuntu0.4) ...
Setting up python-pip-whl (8.1.1-2ubuntu0.4) ...
Setting up python-pip (8.1.1-2ubuntu0.4) ...
0.18.0
[... 381 lines cut ...]

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Starting function 'catkin_lint'
catkin_lint: checked 1 packages and found 0 problems
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'catkin_lint' returned with code '0' after 0 min 0 sec

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Starting function 'after_script'
Testing /root/src/mojin_calibration_data/cob4-12/calibration/calibration_offset.urdf.xacro
Testing /root/src/mojin_calibration_data/cob4-14/calibration/calibration_offset.urdf.xacro
Testing /root/src/mojin_calibration_data/cob4-15/calibration/calibration_offset.urdf.xacro
Testing /root/src/mojin_calibration_data/cob4-17/calibration/calibration_offset.urdf.xacro
Testing /root/src/mojin_calibration_data/cob4-2/calibration/calibration_offset.urdf.xacro
Testing /root/src/mojin_calibration_data/cob4-21/calibration/calibration_offset.urdf.xacro
Testing /root/src/mojin_calibration_data/cob4-22/calibration/calibration_offset.urdf.xacro
Testing /root/src/mojin_calibration_data/cob4-23/calibration/calibration_offset.urdf.xacro
Testing /root/src/mojin_calibration_data/cob4/calibration/calibration_offset.urdf.xacro
Testing /root/src/mojin_calibration_data/mrl-a/calibration/calibration_offset.urdf.xacro
Testing /root/src/mojin_calibration_data/mrl-b/calibration/calibration_offset.urdf.xacro
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'after_script' returned with code '0' after 0 min 2 sec
All tests succeeded
( | )fxm@alpamayo:~/projects/default/robot_ws/src/mojin_calibration_data (rosenv_after_script)$
```

ipa-mdl, 2020-01-18:
> any idea?

`run_ci` passes all environment variables, `run_travis` uses a whitelist. (#277)

It [looks like](https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/docker.sh#L106) `HOME` needs to be added to the [whitelist](https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/scripts/run_travis#L179).

Please append `-- HOME` to your `run_travis` command.
If this works, we can update the whitelist.

fmessmer, 2020-01-18:
thanks, seems to work :point_right: https://github.com/ros-industrial/industrial_ci/pull/466

closed 2020-01-18 by ipa-mdl