ros-industrial/industrial_ci #475: Cannot find rosdep key located in upstream when using colcon on eloquent.

Levi-Armstrong, 2020-02-07:
Not sure where the issues is but it appears that during CI upstream workspace are not correctly being extended. I have created a test [repository](https://travis-ci.com/Levi-Armstrong/ROSDepISSUE/jobs/284433743) so you may recreate the issue. 

To give an overview, the repository depends on another repository which contains a pure cmake package and does contain a package.xml. The repository is listed in the .rosinstall, but I have tested the following locally and rosdep does not have an issue. 

I created two workspace one upstream to contain those package in the .rosinstall and one for the repository in the provided link. I first built the upstream workspace using colcon, then sourced this upstream workspace, followed by the command `rosdep install -q --from-paths "path_to_provided_workspace_src_directory" --ignore-src` giving it the path to the workspace containing the repository in the provided link and everything works as expected. Not sure why CI is having an issue.

gavanderhoorn, 2020-02-07:
@Levi-Armstrong: I changed the title, as I believe it now conveys more concisely the problem you are encountering.

ipa-mdl, 2020-02-07:
This might be a regression of https://github.com/ros-industrial/industrial_ci/pull/454, 5a3936f0fe60ba42485d1cba8a51e231f8294b78 in particular.

@Levi-Armstrong: Please try https://github.com/ipa-mdl/industrial_ci/tree/revert-revert-colcon-chain

Levi-Armstrong, 2020-02-07:
@gavanderhoorn Thank you. 

@ipa-mdl The branch you provided works [here](https://travis-ci.com/Levi-Armstrong/ROSDepISSUE/builds/147931824) is the ci build that passed. Thank you for looking into this.

ipa-mdl, 2020-02-07:
I will need to have a detailed look again, why the current solution is broken (again).

> mpowelson/bionic-qt-ros:eloquent

This image might bundle an older version of the build system (?)

Levi-Armstrong, 2020-02-07:
> mpowelson/bionic-qt-ros:eloquent

The image was just built earlier this week.

mortenfyhn, 2020-03-23:
I'll confirm that I have the same issues in one of my repos, where a ros package pulled in by rosinstall breaks the build with `Cannot locate rosdep definition for [<the package>]`. This is when cloning Industrial CI at master.

If I instead clone https://github.com/ipa-mdl/industrial_ci/tree/revert-revert-colcon-chain as you suggest, it seems to build fine.

Levi-Armstrong, 2020-03-25:
The reverted branch solves the issue shown [here](https://travis-ci.com/github/Levi-Armstrong/ROSDepISSUE/builds/147931824).

ipa-mdl, 2020-04-07:
I am just preparing a test case for #496 and it looks like it is working without any modification for ROS eloquent.

@mortenfyhn, @Levi-Armstrong: Are you using up-to-date Docker images?

ipa-mdl, 2020-04-07:
My test case: https://travis-ci.org/github/ipa-mdl/industrial_ci/jobs/672087673

miguelprada, 2020-05-15:
A replica of your test case but using a [chain of ROS1 workspaces](https://travis-ci.org/github/miguelprada/industrial_ci/jobs/687421041) fails to resolve the dependencies properly.

I am trying to replicate the problem by setting up two workspaces and running rosdep locally but I'm failing to do so (i.e. rosdep succeeds ignoring the package in the upstream workspace).

Any hints about where I can dig further?

ipa-mdl, 2020-05-15:
> Any hints about where I can dig further?

I think I stumbled upon such errors in the past..
If I am not mistaken, `rospack` (older distros) or `ros_environment` (newer distros) is missing.
Without it `ROS_PACKAGE_PATH` does not get [populated](https://github.com/ros/ros_environment/blob/noetic/env-hooks/1.ros_package_path.sh.em).
To test if this is the problem, please try `ADDITIONAL_DEBS="ros-melodic-ros-environment"`

We (I?) removed it from our default installation at some point because it pulls (pulled?) in many dependencies by default.
That's why I [proposed](https://github.com/ros/ros/issues/156) to create the new package ;)

I think we should include it again (in docker.sh), perhaps with `--ignore-missing` in separate line to make it work with older distros as well (without pulling in rospack)

ipa-mdl, 2020-05-15:
Regular packages normally depends on it implicitly, it is just a problem with these empty test packages.

> I think we should include it again (in docker.sh)

or in `ici_setup_rosdep`

miguelprada, 2020-05-15:
You're right. Forcing installation of `ros_environment` [does fix that test case](https://travis-ci.org/github/miguelprada/industrial_ci/jobs/687495558).

> I think we should include it again (in docker.sh), perhaps with `--ignore-missing` in separate line to make it work with older distros as well (without pulling in rospack)

Not sure I follow you here. Are you suggesting to add another apt line (e.g. [here](https://github.com/ros-industrial/industrial_ci/blob/52ac2957708bf7dedbf85b99af7235e48e56717b/industrial_ci/src/docker.sh#L247)) to install `ros-$ROS_DISTRO-ros-environment`?

ipa-mdl, 2020-05-15:
> Are you suggesting to add another apt line (e.g. here) to install ros-$ROS_DISTRO-ros-environment?

Yes, but it should be allowed to fail, because indigo and jade not have this package available.

As an alternative it could be added [here](https://github.com/ros-industrial/industrial_ci/blob/52ac2957708bf7dedbf85b99af7235e48e56717b/industrial_ci/src/workspace.sh#L214).

fmessmer, 2020-06-06:
I can confirm this issue with a setup similar to [`cob_calibration_data`](https://github.com/ipa320/cob_calibration_data) which has the unitary repo [`cob_supported_robots`](https://github.com/ipa320/cob_supported_robots) in it's `UPSTREAM_WORKSPACE` via `rosinstall` file - except for in my setup `mojin_calibration_data` and `mojin_supported_robots` are **private** and thus not released

This is observed with ROS1 distros `kinetic` and `melodic` using `catkin-tools` - other distros/build-tools are not tested

I can confirm that [`revert-revert-colcon-chain`](https://github.com/ipa-mdl/industrial_ci/tree/revert-revert-colcon-chain) (rebased onto latest `master` branch) works for me, too

I was also able to work around it by replacing `UPSTREAM_WORKSPACE=file` with `TARGET_WORKSPACE='. .travis.rosinstall'`

ipa-mdl, 2020-06-07:
> what is the progress of #496?

[My verdict](https://github.com/ros-industrial/industrial_ci/issues/475#issuecomment-610383402) was that it should work without it.

You are hitting the same problem as https://github.com/ros-industrial/industrial_ci/issues/475#issuecomment-629232121
I will prepare a fix for it.

ipa-mdl, 2020-06-07:
I have now merged #522, which installs `ros_environment` (new distros) or `roslib` (old distros).

@fmessmer: please try again with current master

fmessmer, 2020-06-07:
jap, latest `master` works for my usecase - thanks!

Levi-Armstrong, 2020-06-08:
@ipa-mdl This appears to be resolved for ros1 but I am still have an issues with ros2 builds shown [here](https://github.com/ros-industrial-consortium/tesseract_ros2/runs/749886845?check_suite_focus=true).

Levi-Armstrong, 2020-06-08:
Though when we were using your revert branch everything was working. Not sure what was different between what was merged in and your revert branch we were using.

ipa-mdl, 2020-06-08:
@Levi-Armstrong: Thanks for reporting back, I will have a look.

ipa-mdl, 2020-06-08:
@Levi-Armstrong: Your Docker image seems to be outdated..

ipa-mdl, 2020-06-08:
#496 seems to be needed for this special case with cmake-only packages

Levi-Armstrong, 2020-06-09:
@ipa-mdl Thank you for the quick response and resolution. If I understand correctly you took the CMAKE_PREFIX_PATH and appended this to the ROS_PACKAGE_PATH.  Do you think rosdep could do something similar though I am not sure this solves the issue if the pure cmake was installed using the ROS build farm.

ipa-mdl, 2020-06-09:
> Do you think rosdep could do something similar? 

AFAIK it was decided to use index to speedup the lookup.
https://github.com/ros-infrastructure/rosdep/issues/724#issuecomment-641037499 seems to be a good solution.
If it works, I will revert #496 (again).

ipa-mdl, 2020-06-09:
The issue should be resolved for now, but I will keep tracking the discussions in the other issues.

Levi-Armstrong, 2020-06-12:
@ipa-mdl We found a solution outlined in rosdep issue 724. I have not tested with CI but I have confirmed that the pure cmake package gets added the environment variable AMENT_PREFIX_PATH which should solve the rosdep issue.

ipa-mdl, 2020-06-13:
@Levi-Armstrong: That's great to hear! I'd like to revert #496 as soon as we have a proper way to build cmake-only ament packages :)

stertingen, 2021-02-24:
This still seems to be an issue for colcon on ROS 1 (default on noetic). I get this error wth a CMake-only package in the upstream workspace which is needed by a package in the target workspace.

Note:
I tried changing https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/workspace.sh#L293 to
```bash
if [ "$BUILDER" = "colcon" ]; then
```
but this does not seem to resolve the issue.

ipa-mdl, 2021-02-24:
> but this does not seem to resolve the issue.

I am not sure, if the ` CMAKE_PREFIX_PATH`  trick works with the catkin build system.

> EDIT: As workaround, we could set catkin_tools as default for noetic. Works fine for me.

I don't want to change the default too easily, but this would be the last resort..

ipa-mdl, 2021-02-24:
I think the reason is: as https://github.com/ros-industrial/industrial_ci/issues/624#issuecomment-779493671
`colcon` does not use the `catkin` build system for `setup.bash`.

ipa-mdl, 2021-02-24:
>I am not sure, if the CMAKE_PREFIX_PATH trick works with the catkin build system.

For ROS1 `/share` needs to be appended to each path.

gavanderhoorn, 2021-02-24:
Related Q&A on ROS Answers: [How to use rosdep to install dependencies of a ROS 1 install folder (built by Colcon)](https://answers.ros.org/question/369324).

ipa-mdl, 2021-02-24:
> EDIT: As workaround, we could set catkin_tools as default for noetic. Works fine for me.

@stertingen: I tried it with a dummy package and it did not work for me.

ipa-mdl, 2021-02-24:
>  tried it with a dummy package and it did not work for me.

I got it working now..
It just works by coincidence, because `catkin_tools` uses a merged install space by default.
With `--isolate-install` (colcon's default) it does not work anymore.

However, this case should be fixable by sourcing  `/opt/ros/noetic/etc/catkin/profile.d/1.ros_package_path.sh`.

ipa-mdl, 2021-02-24:
> However, this case should be fixable by sourcing /opt/ros/noetic/etc/catkin/profile.d/1.ros_package_path.sh.

No, this does not work.
Apparently, there is no (correct) way to "register" a cmake-only package with ROS1 (ROS2 has ament index).
This means as well that you cannot "depend" on it in package.xml.
However, for the sake of usability, we can register all `CMAKE_PREFIX_PATH` as ROS packages and not only catkin packages.

ipa-mdl, 2021-02-24:
One option (colcon only) would be to register the package in a hook (https://colcon.readthedocs.io/en/released/user/configuration.html#colcon-pkg-files).

gavanderhoorn, 2021-02-24:
> Apparently, there is no (correct) way to "register" a cmake-only package with ROS1

well, anything with a `package.xml` will be considered a 'ROS package'.

that's why many plain-CMake pkgs install a manifest.

Cartographer is a famous example fi.

Levi-Armstrong, 2021-02-24:
To solve this issue I added this to all plain-CMake packages

``` cmake
  # Allows Colcon to find non-Ament packages when using workspace underlays
  file(WRITE ${CMAKE_CURRENT_BINARY_DIR}/share/ament_index/resource_index/packages/${PROJECT_NAME} "")
  install(FILES ${CMAKE_CURRENT_BINARY_DIR}/share/ament_index/resource_index/packages/${PROJECT_NAME} DESTINATION share/ament_index/resource_index/packages)
  file(WRITE ${CMAKE_CURRENT_BINARY_DIR}/share/${PROJECT_NAME}/hook/ament_prefix_path.dsv "prepend-non-duplicate;AMENT_PREFIX_PATH;")
  install(FILES ${CMAKE_CURRENT_BINARY_DIR}/share/${PROJECT_NAME}/hook/ament_prefix_path.dsv DESTINATION share/${PROJECT_NAME}/hook)
  file(WRITE ${CMAKE_CURRENT_BINARY_DIR}/share/${PROJECT_NAME}/hook/ros_package_path.dsv "prepend-non-duplicate;ROS_PACKAGE_PATH;")
  install(FILES ${CMAKE_CURRENT_BINARY_DIR}/share/${PROJECT_NAME}/hook/ros_package_path.dsv DESTINATION share/${PROJECT_NAME}/hook)
```

closed 2021-02-24 by Levi-Armstrong

ipa-mdl, 2021-02-24:
> well, anything with a package.xml will be considered a 'ROS package'. That's a requirement for CMake projects which want to be buildable with Catkin.

@gavanderhoorn: I agree, but [ros_enviroment](https://github.com/ros/ros_environment/blob/noetic/env-hooks/1.ros_package_path.sh.em#L9) does not.

-> ROS1 packages are detected by (rospack/rosdep), if they install a `package.xml` in a `share` folder, with `.catkin` file next to it ;)
For merged install spaces the latter is always present, but for isolated install spaces..

ipa-mdl, 2021-02-24:
> To solve this issue I added this to all plain-CMake packages

@Levi-Armstrong: Thanks for sharing this snippet!

Today I learned that this 
```cmake
  file(WRITE ${CMAKE_CURRENT_BINARY_DIR}/share/${PROJECT_NAME}/hook/ros_package_path.dsv "prepend-non-duplicate;ROS_PACKAGE_PATH;")
  install(FILES ${CMAKE_CURRENT_BINARY_DIR}/share/${PROJECT_NAME}/hook/ros_package_path.dsv DESTINATION share/${PROJECT_NAME}/hook)
```
could be done in one step:

```cmake
  install(CODE "file(WRITE \"${CMAKE_INSTALL_PREFIX}/share/${PROJECT_NAME}/hook/ros_package_path.dsv\" \"prepend-non-duplicate;ROS_PACKAGE_PATH;\")")
```
Not sure, if this is better, though.

gavanderhoorn, 2021-02-24:
> @gavanderhoorn: I agree, but [ros_enviroment](https://github.com/ros/ros_environment/blob/noetic/env-hooks/1.ros_package_path.sh.em#L9) does not.
> 
> -> ROS1 packages are detected by (rospack/rosdep), if they install a `package.xml` in a `share` folder of a `CMAKE_PREFIX_PATH` entry with a `.catkin` file ;)
> For merged install spaces the latter is always present, but for isolated install spaces..

that's interesting.

I've literally never run into any problems with plain-CMake pkgs installing manifests, but perhaps:

 1. Catkin 'secretly' always just installed those `.catkin` markers for me
 2. I've *never* used `install` spaces

the latter is probably not true ..

PR #496 (2020-06-08): work-around for chained colcon workspaces with cmake-only packages
This reverts commit 5a3936f0fe60ba42485d1cba8a51e231f8294b78.
Fixes #475 

ToDo:

- [ ] add test case
- [x] cherry-pick changes from https://github.com/ros-industrial/industrial_ci/pull/454