ros-industrial/industrial_ci #476: Appropriate way of using rosdep and package.xml depend tags for use with industrial_ci

doisyg, 2020-02-13:
Hello,
Complete beginner in the setup of a CI workflow here. industrial_ci looks like a good entry point. I am starting using Github Actions with the .github/workflows/main.yml described in the documentation.
I tried with a simple workspace: only one independent package and it works well.
However when trying with a more complex workspace where I have:

`package_A`
`package_B`

with `package_B` depending on `package_A` and hence have `<depend>package_A</depend> `declared in its `package.xml`, the CI rosdep step fails with:
```
  Starting function 'install_target_dependencies'
  ERROR: the following packages/stacks could not have their rosdep keys resolved
  to system dependencies:
  package_B: Cannot locate rosdep definition for [package_A]
```

What should be the proper approach? skipping `package_A` key with `ROSDEP_SKIP_KEYS` ? Or maybe I have been using for all these years incorrectly the package.xml tags? On my local machine, rosdep succeed when I run ` rosdep install --from-paths src --ignore-src -y`. How can I know what is the rosdep command run by industrial_ci? Should I fork this repo and customize it ?

Thanks!

ipa-mdl, 2020-02-14:
Normally, this should not be an issue. You don't have to (and should not) set `ROSDEP_SKIP_KEYS´.

Are `package_A`  and `package_B` in one repository?

doisyg, 2020-02-17:
Good catch, `package_A` is a submodule of the main repo and stored in another repo (but same account). 
I could not reproduce the issue when both pacakges are in the same repo.
So I guess the issue comes from the initialization of the submodule. Is it suppose to be done automatically or a step should be added?
(changing title of the issue)

doisyg, 2020-02-17:
Minimal example with submodule which fails:
https://github.com/doisyg/test_CICD

ipa-mdl, 2020-02-17:
Submodules don't get checked out by default (?).
Please try https://github.com/actions/checkout#checkout-submodules

doisyg, 2020-02-18:
Yes, thanks, working by using https://github.com/actions/checkout#checkout-submodules

closed 2020-04-08 by ipa-mdl