ros-industrial/industrial_ci #479: Private Customer Docker Images

tianshiz, 2020-02-19:
I'm trying to use a custom docker image(uploaded to docker hub) with ros and all my dependencies pre-installed with ROS industrial CI. The following is my bitbucket pipelines. I've set the ENV DOCKER_IMAGE to "username\repo" and now it does pull the repo. However, I would like to make my repo private. How can I assign credentials or access tokens?

```
image: docker:git
    pipelines:
      default:
         - step:
             services:
               - docker
             caches:
               - docker
             script:
               - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
               - .industrial_ci/bitbucket.sh ROS_DISTRO=kinetic

definitions:
  services:
    docker:
      memory: 2048
```

ipa-mdl, 2020-02-20:
According to the [Bitbucket documentation](https://confluence.atlassian.com/bitbucket/run-docker-commands-in-bitbucket-pipelines-879254331.html), just login first:
> docker login --username $DOCKER_USERNAME --password $DOCKER_PASSWORD

To set the credentials, use [(secured) variables](https://confluence.atlassian.com/bitbucket/variables-in-pipelines-794502608.html).

Instead of your actual password, better use a [personal access token](https://www.docker.com/blog/docker-hub-new-personal-access-tokens/) if possible.

tianshiz, 2020-02-20:
thanks, that was it!

closed 2020-02-20 by tianshiz