ros-industrial/industrial_ci #486: Add version tags to specify version from GitHub Actions

Tiryoh, 2020-03-18:
The documentation says to use the master branch for GitHub Actions.
https://github.com/ros-industrial/industrial_ci/tree/master#for-github-actions

The official documentation of GitHub Actions recommends using a version tag.
https://github.com/actions/toolkit/blob/6459481e988198bf53e186abf23350b085b1250f/docs/action-versioning.md#compatibility

What do you think about adding version tags to the master branch as it has been added to the legacy branch before?

ipa-mdl, 2020-03-22:
> What do you think about adding version tags to the master branch as it has been added to the legacy branch before?

Yes, it is about time.. @130s: do you agree?

 > The official documentation of GitHub Actions recommends using a version tag.

This might make sense in general (given a stabe ecosystem), but for `industrial_ci` we  recommend branches, because the ROS ecosystem is quite dynamic and sometimes there are upstream (repository keys, rosdep, buildfarm, etc) breaks.

Using the branch approach, we have been able to deploy hotfixes etc. to all users on short notice.

I don't see the need to change this for Github Actions..
Did you face any problems?

130s, 2020-03-30:
> > What do you think about adding version tags to the master branch as it has been added to the legacy branch before?
> 
> Yes, it is about time.. @130s: do you agree?

Would either @ipa-mdl or @Tiryoh mind clarifying the suggested direction? I kind of know what it means, but don't want to make a false guess.

Responding to other parts in the meantime,

> The official documentation of GitHub Actions recommends using a version tag. https://github.com/actions/toolkit/blob/6459481e988198bf53e186abf23350b085b1250f/docs/action-versioning.md#compatibility

I think the example in the link uses `master` branch as the bleeding edge, which is the same as `industrial_ci` repo. So the suggesting to switch to `tag` is reasonable. With `master` branch on this repo I think stability is far from being promised (despite this is the community-powered repo) because it's the latest development. However, looking at how quickly we've been responding to bug reports (nearly 100% credit goes to @ipa-mdl?), this model has been working, quite well IMO, so far.

Users repo pointing to `tag` would also change their workflow -- Per each release, users would have to update their repo to point to the new tag, which is similar to what I disliked the idea of `git submodule` early on when I started making this repo. I/we wanted to free users from CI maintenance as much as possible.

In case the current flow stopped working, then we can take a different approach. Maybe create a stable branch and recommend users to point that branch.

ipa-mdl, 2020-04-07:
@130s:
I just meant that it is time about to add a new version tag (0.9.0?) for the master branch as well.
I have added this task to #493 as well.

ipa-mdl, 2020-04-08:
I will add tags later today (#418), but I advise against using tags `with industrial_ci`.

closed 2020-04-08 by ipa-mdl

Tiryoh, 2020-04-08:
Sorry for the late reply.  I had overlooked the notification.

> adding version tags to the master branch as it has been added to the legacy branch before

I meant adding tags like this(https://github.com/ros-industrial/industrial_ci/releases/tag/0.8.0), as @ipa-mdl has mentioned here(https://github.com/ros-industrial/industrial_ci/issues/486#issuecomment-610299460).

> This might make sense in general (given a stabe ecosystem), but for industrial_ci we recommend branches, because the ROS ecosystem is quite dynamic and sometimes there are upstream (repository keys, [...]

> Users repo pointing to tag would also change their workflow -- Per each release, users would have to update their repo to point to the new tag, which is similar to what I disliked the idea of git [...]

I hadn't thought that far. I have not face a problem using industrial_ci on GitHub Actions for now.

Looking at the status of the CI in the master branch, it seemed that sometimes an error appeared, so I suggested that it could be more stable to use industrial_ci by specifying the version. 

Thanks to your explanation, it seems reasonable to point to the master branch on GitHub Actions.