ros-industrial/industrial_ci #487: Support Python formatting with Black

mortenfyhn, 2020-03-20:
Hi yall and thank you for making this wonderful tool. I just recently discovered it (after having previously overlooked it because I was confused by the name). I'm testing it out now.

One thing I like to run in my pipelines is [Black](https://black.readthedocs.io/en/stable/) for Python code formatting. I see you have clang-format already supported. Are you open to supporting Black as well? I could probably attempt a PR.

ipa-mdl, 2020-03-22:
> Are you open to supporting Black as well?

Yes, definitely.

>  I could probably attempt a PR.

If you do not need a ROS environment (I don't see why black would need one), it should work the same as for `clang-format`.
It might even make sense to have a common codebase.

mortenfyhn, 2020-03-22:
Cool. Yeah, no ROS environment should be needed. I'm already running black checks outside of Industrial CI, before setting up and building any ROS stuff.

PR #490 (2020-12-16): Add support for Python style check with Black
Fix #487.

This seems to work. I've made a test repo where I run this branch of my fork of Industrial CI: https://github.com/mortenfyhn/rospy_test (Using Semaphore CI, see .semaphore/semaphore.yml)

Here is the output of the passing master branch build: https://mortenfyhn.semaphoreci.com/workflows/b69d6f0f-afca-421f-8d22-3aa1b2d829f1

Here is the output of a build where I deliberately mis-formatted a Python file: https://mortenfyhn.semaphoreci.com/workflows/ef3c8bda-8838-4420-99ff-adc1464f4ef1

I mostly just copied and modified `clang_format_check. [... cut]
closed 2020-12-16 by ipa-mdl