ros-industrial/industrial_ci #488: Error using Gitlab docker runner?

rarrais, 2020-03-20:
Hi!

I am currently having trouble running the industrial_ci on GitLab using the docker runner, which I did with success in the past (some months ago). The issue seems to manifest on my organization private Gitlab as well as on Gitlab.com. 

Please check [this ](https://gitlab.com/rarrais/ros-ci) for an example repository where this issue manifests. 

There are two pipelines, both referencing the same commit and using the same .gitlab-ci.yml file, however, the [first one](https://gitlab.com/rarrais/ros-ci/-/jobs/479803249) is succeeded without problems, using Gitlab's own runners' infrastructure, which seems to be 'docker+machine'. The [second pipeline](https://gitlab.com/rarrais/ros-ci/-/jobs/479816704) (same commit) fails while using a 'Specific Runner', using the 'docker' runner kind. Maybe this issue is similar to issue #478, thus the addition of the workaround suggested to the yml file, but without success. 

For easiness, reproducing here the apparent error message:

```
RUN sed -i "/^# deb.*multiverse/ s/^# //" /etc/apt/sources.list     && apt-get update -qq     && apt-get -qq install --no-install-recommends -y         build-essential         python-catkin-tools         python-pip         python-rosdep         python-wstool         ros-kinetic-catkin         ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*
 time="2020-03-20T17:11:23Z" level=error msg="failed to dial gRPC: cannot connect to the Docker daemon. Is 'docker daemon' running on this host?: dial tcp: lookup docker on 172.31.0.2:53: no such host"
 error during connect: Post http://docker:2375/v1.40/build?buildargs=%7B%7D&cachefrom=%5B%5D&cgroupparent=&cpuperiod=0&cpuquota=0&cpusetcpus=&cpusetmems=&cpushares=0&dockerfile=Dockerfile&labels=%7B%7D&memory=0&memswap=0&networkmode=default&pull=1&rm=1&session=x80yjkq8l5s5gbbfobrauk2br&shmsize=0&t=industrial-ci%2Fkinetic%2Fubuntu%3Axenial&target=&ulimits=null&version=1: context canceled 
```

Thank you in advance for the help and appreciate the effort put into this repository!

ipa-mdl, 2020-03-22:
For now I would say that this is a problem with your Docker-based runner.
The main error seems to be `dial tcp: lookup docker on 172.31.0.2:53: no such host"`
Port 53 is used for DNS queries. I am not sure why the hostname "docker" is used.
Perhaps you have to use an older version of the `docker:*`  images, e.g. to match your host's version.

ThijsRay, 2020-09-16:
I have come across a similar same problem. I have tried to run `gitlab-runner exec docker melodic` on two different machines in two different networks (one `Arch Linux`, one `Debian` machine), but the runner halts at the exact same point. `time="2020-09-16T08:15:14Z" level=error msg="failed to dial gRPC: cannot connect to the Docker daemon. Is 'docker daemon' running on this host?: dial tcp: lookup docker on 10.0.0.1:53: no such host"`. Any suggestions on how to proceed?

ipa-mdl, 2020-12-13:
This does not seem to be related to `industrial_ci`, so I will close it.

closed 2020-12-13 by ipa-mdl