ros-industrial/industrial_ci #492: How to blacklist some packages in catking build

ykhedar, 2020-03-31:
Hi All, I am just starting to use the industrial_ci for our project:   
https://github.com/laxnpander/OpenREALM

insudtrial_ci is a wonderful project i must say. I have one doubt though: 

How and where can i specify to blacklist a package for catkin config.

I went throuth the following Pull Request (https://github.com/ros-industrial/industrial_ci/pull/361) from @ipa-mdl in order to find a clue since this issue was mentioned somewhere (https://github.com/ros-industrial/industrial_ci/issues/172): 

but could not find a clue.

I would like to know if its possible to blacklist packages easily or one needs to do some hacks? 

Thanks in advance. :) and once again great work on making this tool available to the public.
khedar

ipa-mdl, 2020-04-07:
Are you using the `legacy` or the `master` version?

In the `legacy` version there is no blacklist, only `BUILD_PKGS_WHITELIST`.
In contrast, the `master` version does not have a whitelist, but allows to remove certain folders before the build: https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#workspace-management

ykhedar, 2020-04-07:
Thanks for your repsonse. 

I am using the ```master``` branch.

I tried the ```TARGET_WORKSPACE='. -pkg_a'``` tag with filtering out the blacklist package. But the issue is one other package depends on it in its package.xml files. So even though the ```TARGET_WORKSPACE``` allows me to remove the blacklisted package, the build fails. 

Is there no way to execute ```catkin config --blacklist pkg_a``` before the target workspace is built?

One hack which i could think of is to remove the blacklist package from the package.xml of the node depending on this blacklist package using ```sed```, not sure how and where would i need to execute it though.

Thanks again,
khedar

ipa-mdl, 2020-04-07:
> I tried the TARGET_WORKSPACE='. -pkg_a' tag with filtering out the blacklist package. But the issue is one other package depends on it in its package.xml files

Why do you want to blacklist a package, which is needed by a different package?
You would have to blacklist/remove both.

> Is there no way to execute catkin config --blacklist pkg_a before the target workspace is built?

There is a way using hooks but, I don't recommend it.
The rationale behind is that `colcon` should be used later on.
`catkin-tools` has never been an official ROS build tool.

According to the `catkin-tools` documentation, `--blacklist` should not prevent dependent packages from being built. Perhaps this a hidden feature..

ipa-mdl, 2020-04-07:
I think, in your [special case](https://travis-ci.org/github/laxnpander/OpenREALM/jobs/669790822), you have to set `ROSDEP_SKIP_KEYS=psl` as well.

ykhedar, 2020-04-07:
Thanks for the suggestion. It seems to solve the problem now. 

> Why do you want to blacklist a package, which is needed by a different package?
You would have to blacklist/remove both.

We have one node which depends on CUDA. This node is an implementation of the base node class. Other nodes also depend on the base node and not the implementation node directly. So if i remove the base node as well with the CUDA dependent implementation node then i have further errors from other which depend on the base node now.  

With the help of Cmake arguments we can remove this package from compiling but the catkin build system on the travis (maybe also locally on my computer) also reads the package.xml where this dependency is mentioned. 

On a local computer we specify both in the catkin config and also as a CMAKE argument to blacklist this package and then it works. 

Maybe we need to think about the dependenies once again.

But thanks again for your quick response. I would close this issue for now.

closed 2020-04-07 by ykhedar

ipa-mdl, 2020-04-07:
> also reads the package.xml where this dependency is mentioned.

Yes, ` rosdep`  is reading it. You can bypass it by setting `ROSDEP_SKIP_KEYS=psl`

> We have one node which depends on CUDA. This node is an implementation of the base node class. Other nodes also depend on the base node and not the implementation node directly.

Normally, the base class should not depend on an implementation.
I don't know if you plan to release the packages at some point, but ` bloom` uses `package.xml` as well to specify the dependencies between the binary packages.