ros-industrial/industrial_ci #494: Make master branch the default

ipa-mdl, 2020-04-07:
I would propose to make the switch until end of June.
This should be announced in #493 as well.

@130s, @miguelprada, @gavanderhoorn: Any objections?

gavanderhoorn, 2020-04-07:
June seems pretty far away.

We've had a grace period of many months (a year perhaps?) already.

End of May would seem sufficient to me, but I'm not in a hurry, and if you prefer end of June that would also be fine.

I would recommend to be vocal about it. So Discourse, the tracker here (perhaps this issue is already sufficient), and, maybe, also have `legacy` branch print a notice in the logs. Yes, "no one" reads those as long as everything is green, but it might still pay off if/when someone happens to *do* read the log for whatever reason.

ipa-mdl, 2020-04-07:
> June seems pretty far away.

I am okay with "in June / after May" as well..
(-> 2 months of grace period after the announcement)

> maybe, also have legacy branch print a notice in the logs

Yes, this should get added to #397

ipa-mdl, 2020-04-07:
> Yes, this should get added to #397

I have just decided against this..
Instead, I will create a "transitional" branch, which will show a warning.
This branch can be deleted safeky after master is default again.

ipa-mdl, 2020-04-07:
I have updated the schedule.
The switch will happen on June 8. One year after the [announcement](https://discourse.ros.org/t/industrial-ci-migration-needed-for-recent-and-future-changes/9462)

ipa-mdl, 2020-06-08:
switched to `master_transitional`

ipa-mdl, 2020-06-30:
switched to `master` :tada:

closed 2020-06-30 by ipa-mdl

gavanderhoorn, 2020-06-30:
Now to see how many CI setups get broken ;)