ros-industrial/industrial_ci #502: Log is "null" after switching to master branch

wxmerkt, 2020-04-18:
Hi,
We use industrial_ci on Travis and I am checking if I can migrate our CI from `legacy` to `master`. In a new PR, where I change the default branch, all the logs on Travis are `null`. 

- PR changing branches: https://github.com/ipab-slmc/exotica/pull/705/files#diff-354f30a63fb0907d4ad57269548329e3R40
- Build using master branch with empty log: https://travis-ci.org/github/ipab-slmc/exotica/jobs/676718935
- Empty log: https://api.travis-ci.org/v3/job/676718935/log.txt
- Build with legacy branch with log: https://travis-ci.org/github/ipab-slmc/exotica/jobs/676719431

Have you seen this issue before or is there a work-around?

Thanks,
Wolfgang

ipa-mdl, 2020-04-18:
That's really strange!
For now, I doubt that it is a problem with switching the branches.
Looks like Travis CI has a problem..

Even your legacy job has a 'null' log now: https://api.travis-ci.org/v3/job/676719431/log.txt

I just restarted [one of my jobs](https://travis-ci.org/github/ipa-mdl/industrial_ci/jobs/672063165) and now the log is gone as well.

ipa-mdl, 2020-04-18:
It looks like only travis-ci.org is affected, travis-ci.com works fine.

ipa-mdl, 2020-04-19:
Travis CI seems to have recovered.. https://api.travis-ci.org/v3/job/676718935/log.txt is not empty anymore.

closed 2020-04-19 by ipa-mdl

wxmerkt, 2020-04-19:
Thanks for looking into this and happy to hear it's resolved :-).