ros-industrial/industrial_ci #505: Foxy nightly rosdep update cannot find distro

Briancbn, 2020-05-06:
### Bug Description
During my **foxy** nightly [build](https://travis-ci.com/github/Briancbn/packml_ros2/jobs/328472167), 

`rosdep update` failed with the message 
```
Requested distribution "foxy" is not in the index.
```
[Here](https://github.com/Briancbn/packml_ros2/blob/62e7c7423b4b49c91f0a21c9858b9fa01006eefc/.travis.yml) is my `.travis-ci.yml` configuration

### Proposed Solution

I did some investigation, it seems that after #471 is merged, the nightly build will have this error .The environment variable `ROS_DISTRO` is foxy, but it is not really a ros distro released . Should we temporarily treat **foxy** as an exception here, or just revert the original PR for the moment.

ipa-mdl, 2020-05-06:
Yes, it looks like the nightly image does not support `rosdep` install for ROS packages (https://github.com/osrf/docker_images/pull/307)

>  Should we temporarily treat foxy as an exception here, or just revert the original PR for the moment.

I would propose something in the middle: introduce a flag (like ` ROS_VERSION_NIGHTLY`), which bypasses #471 
We can switch `foxy` soon to a proper setup, but nightly builds should stay possible somehow, one idea was to implement `ROS_DISTRO=nightly` later

ipa-mdl, 2020-05-06:
I could work on this tonight, but not now..

Briancbn, 2020-05-06:
Thanks @ipa-mdl!
> I would propose something in the middle: introduce a flag (like `ROS_VERSION_NIGHTLY`), which bypasses #471
> We can switch `foxy` soon to a proper setup, but nightly builds should stay possible somehow, one idea was to implement `ROS_DISTRO=nightly` later

Both sounds good!

ipa-mdl, 2020-05-06:
I merged the fix for rosdep update (#506).

In addition, #507 switches to the ROS repositories.
Not sure, when to merge this..

Briancbn, 2020-05-08:
Thanks @ipa-mdl, it is working for me. The `ROSDISTRO_INDEX_URL` is certainly a better fix!

closed 2020-05-08 by Briancbn