ros-industrial/industrial_ci #510: "ROS distro 'noetic' is not supported"

130s, 2020-05-14:
E.g. https://travis-ci.org/github/ros/executive_smach/jobs/686410025 for https://github.com/ros/executive_smach/pull/71
```
$ git clone https://github.com/ros-industrial/industrial_ci.git .ci_config
0.01s$ .ci_config/travis.sh
'USE_DEB' is deprecated. Please migrate to UPSTREAM_WORKSPACE.
ROS distro 'noetic' is not supported
The command ".ci_config/travis.sh" exited with 1.
```

gavanderhoorn, 2020-05-14:
I believe noetic support was added in #499, but it's only present in `master`.

130s, 2020-05-14:
Seems to be the case before https://github.com/ros-industrial/industrial_ci/issues/494 to be resolved.

gavanderhoorn, 2020-05-14:
Or just `clone -b master` right now.

ipa-mdl, 2020-05-14:
As @gavanderhoorn  already wrote, `noetic` is supported in the `master` branch.
(See [announcement](https://discourse.ros.org/t/industrial-ci-rollout-of-version-0-10-0-with-ros2-github-actions-support/13582) as well).

I will close this issue now, because there is nothing left to do.

closed 2020-05-14 by ipa-mdl