ros-industrial/industrial_ci #511: Job control in master branch

wxmerkt, 2020-05-17:
Hi,
Thank you for creating and maintaining this easy to use CI setup. We have been using Industrial CI for a long time and recently migrated to the master branch using the Migration Guide. However, due to heavily templated code we rely on limiting the number of concurrent/parallel jobs to avoid exhausting memory on Travis (and as a result stalling/timing out the build). From the migration guide, it is clear that this option has been removed for the time being:
> industrial_ci does not set defaults for the number of build jobs anymore. In addition the job-control options got removed for the time being.

Is there an alternate way to specify maximum number of jobs or plans to bring this feature back?

Thanks,
Wolfgang

ipa-mdl, 2020-05-18:
> it is clear that this option has been removed for the time being

Yes, this has been removed.
Mostly because it was not portable.
In addition, the `legacy` defaults to 4 parallel packages with 8 jobs (each?), which is not a reasonable choice for Travis etc.
The `master` branch does not set any default, but uses the buildtool's default (number of cores?)

Travis CI uses machines with [2 cores and 7.5GB memory](https://docs.travis-ci.com/user/reference/overview/#virtualization-environments).

> Is there an alternate way to specify maximum number of jobs

Not really, perhaps by limiting the number of core for Docker..

> plans to bring this feature back

Plans (in general): yes.
Specific ideas: no.

At first we need to define the nature of the limit.
I could imagine a total number of parallel jobs, which should get mapped to the buildtool (catkin-tools, colcon, catkin_make[_isolated]) options somehow

wxmerkt, 2020-05-20:
Thanks for the detailed response. Passing `DOCKER_RUN_OPTS="--cpus=\"1\""` resolved it for us.

ipa-mdl, 2020-06-07:
@wxmerkt: please test #526

wxmerkt, 2020-06-08:
Solved by #526

closed 2020-06-08 by wxmerkt