ros-industrial/industrial_ci #512: [Master] Fails because docker_cp cannot copy to non-existing directory

ipa-jba, 2020-05-25:
Hi,

at the moment our CI-Pipelines (with new master branch) fail because of
```
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
00:53
 Starting function 'prepare_docker_image'
 Building image 'industrial-ci/kinetic/ubuntu:xenial':
 FROM ubuntu:xenial
 ENV ROS_DISTRO kinetic
 ENV LANG C.UTF-8
 ENV LC_ALL C.UTF-8
 RUN echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections
 RUN apt-get update -qq && apt-get -qq install -y apt-utils gnupg2 wget ca-certificates lsb-release dirmngr build-essential
 RUN for i in 1 2 3; do { apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key 'C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654'; } &&  break || sleep 1; done
 RUN echo "deb http://packages.ros.org/ros-testing/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros1-latest.list
 RUN sed -i "/^# deb.*multiverse/ s/^# //" /etc/apt/sources.list  && apt-get update -qq
 <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
 Function 'prepare_docker_image' returned with code '0' after 0 min 52 sec
 destination "77d39af9ef6a5c8e83ad0b2bdcbf08b53727127502baeaf0d73f4b988f687c17:/root/" must be a directory
 ERROR: Job failed: exit code 1
```

Most likely this is because the `/root/` directory does not exist in the newly built container. Right now I'm testing whether a quick fix works (https://github.com/ros-industrial/industrial_ci/compare/master...ipa-jba:mkdir-before-copy) and will comment whether it worked

gavanderhoorn, 2020-05-25:
> Most likely this is because the `/root/` directory does not exist in the newly built container

No `/root` in a Linux container?

ipa-jba, 2020-05-25:
If i start a container with the newest ubuntu:{xenial, bionic, focal} it is there, but all of our (gitlab) pipelines (even using legacy branch) failed with the same error
```
 <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
 Function 'prepare_docker_image' returned with code '0' after 1 min 18 sec
 destination "89ae9fcc1c78b709fbe4c5fa3059dc5bd23554c336aaa09bef6d7aa3b38558b5:/root/" must be a directory
```

I'm not certain what's happening there

gavanderhoorn, 2020-05-25:
Well I'm not saying you're lying, but it would be the first time I see a Linux based container without that directory present.

Were I to come across this problem, I would first start looking at other causes.

ipa-jba, 2020-05-25:
since I ran into another issue: https://gitlab.com/gitlab-org/gitlab-runner/-/issues/25803 It might be coinciding with that one and a problem of the docker daemon /gitlab runner instead of the actual ci-pipeline.

I will try it without the fix and report back

ipa-jba, 2020-05-25:
Reporting back: 

I pinned the "service" from "docker:git" to "docker:18-git" which is the fix for the forementioned issue and this solves all the ci-issues --> close and thank you for your input

gavanderhoorn, 2020-05-25:
Thanks for checking it and reporting back.

Timple, 2020-05-27:
We are having this issue as well. I think this is closed too early.
If "docker:18-git" is the final solution, we need to update the README's and examples.

ipa-mdl, 2020-05-27:
> If "docker:18-git" is the final solution, we need to update the README's and examples.

It is not a fix in general.
This problem is not even related to Gitlab, but a apparently Docker issues: The most recent client does not work with older daemons (https://github.com/docker/cli/pull/2537).
At least on Gitlab.com, `docker:19.03.5-dind` (current industrial_ci config) woks as well

>  I think this is closed too early.

I agree

ipa-mdl, 2020-05-27:
> At least on Gitlab.com, docker:19.03.5-dind (current industrial_ci config) works as well

So [does `docker:dind`](https://gitlab.com/ipa-mdl/industrial_ci/pipelines/150142187)

ipa-mdl, 2020-06-11:
This seems to have been fixed (https://github.com/docker/cli/pull/2538, probably released in 19.03.10).
Please test again with the latest version.

closed 2020-12-13 by ipa-mdl