ros-industrial/industrial_ci #514: Our GitHub workflows suddenly failed with 'No CMAKE_C_COMPILER could be found.'.

ShotaAk, 2020-05-28:
# Describe the bug
Our GitHub workflows suddenly failed with error messages: `No CMAKE_C_COMPILER could be found.` and `    No CMAKE_CXX_COMPILER could be found.`.

# Environments

YAML file:

```yaml
steps:
      - uses: actions/checkout@v2
      - uses: "ros-industrial/industrial_ci@master"
```

More details:
- https://github.com/rt-net/raspimouse2/blob/master/.github/workflows/industrial_ci.yml
- https://github.com/rt-net/rt_usb_9axisimu_driver/blob/master/.github/workflows/industrial_ci.yml

# To Reproduce

1. Open a successful job of your repository.
2. Press `Re-run jobs`.
3. The job failed.

# Full log or Screenshots

```txt
Starting function 'build_target_workspace'
  Starting >>> raspimouse_msgs
  --- stderr: raspimouse_msgs
  CMake Error at CMakeLists.txt:2 (project):
    No CMAKE_C_COMPILER could be found.
  
    Tell CMake where to find the compiler by setting either the environment
    variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
    the compiler, or to the compiler name if it is in the PATH.
  
  
  CMake Error at CMakeLists.txt:2 (project):
    No CMAKE_CXX_COMPILER could be found.
  
    Tell CMake where to find the compiler by setting either the environment
    variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
    to the compiler, or to the compiler name if it is in the PATH.
```

More details:
- https://github.com/rt-net/raspimouse2/runs/715160653?check_suite_focus=true
- https://github.com/rt-net/rt_usb_9axisimu_driver/runs/715152058?check_suite_focus=true

# Additional context

The test for ros-industrial/industrial_ci seems to failed with the same error.
https://travis-ci.com/github/ros-industrial/industrial_ci/jobs/340483108

ipa-mdl, 2020-05-28:
Thanks for reporting!
I noticed the same problem with Travis.
It looks like the official ROS Docker images have changes yesterday and no do not bundle the build essentials anymore.
I have added  #515 to track this and come up with a better fix.

Please confirm if the quickfix in #516 solved your problems as well.

ShotaAk, 2020-05-28:
@ipa-mdl Thank you for your survey and quickfix!

I just re-ran the above workflows that results ROS kinetic/melodic build succeeded but ROS dashing build failed.

Error messeage:
```txt
 Starting function 'install_target_dependencies'
  E: Unable to locate package ros-dashing-rclcpp
  ERROR: the following rosdeps failed to install
    apt: command [apt-get install -y -qq ros-dashing-rclcpp] failed
  executing command [apt-get install -y -qq ros-dashing-rclcpp]
```
https://github.com/rt-net/raspimouse2/runs/716149818

ipa-mdl, 2020-05-28:
Thanks! At least ROS1 works again :)

>   E: Unable to locate package ros-dashing-rclcpp

I will check this, but I cannot work on it right now.

ShotaAk, 2020-05-28:
> I will check this, but I cannot work on it right now.

That's fine! Thanks:+1:

ipa-mdl, 2020-05-28:
@ShotaAk: I found the problem, is unrelated to the other changes..

For your dashing build you set "ROS_REPO=ros", which actually configures ROS1.

The idea was to just use actual the name from `ROS_REPOSITORY_PATH`:
`ros` -> ROS1 repo
`ros2` -> ROS2 repo
`main` -> auto repo

Without #516, the actual value was ignores and the Docker images was used instead.

So the simple fix would be to just change in your config `main`  or ` ros2`.

However, the current behavior might be very confusing (even to me ^^).

` industrial_ci`  should raise an error if the a wrong repo got confgured.
In addtition, a more user-friendly approach would be to make ROS_REPO=`ros` behave like `main`  and add specific `ros1*`  aliases as well.

I implemented the ladder in #517, error checks etc. will be added later

ShotaAk, 2020-05-28:
@ipa-mdl 
> For your dashing build you set "ROS_REPO=ros", which actually configures ROS1.

I didn't realized the wrong parameter. :scream: 
Thank you for checking our repository!

I will fix it to `ROS_REPO=main`. (I don't want to be bothered by ROS and ROS2 anymore. :laughing: )

> I implemented the latter in #517, error checks etc. will be added later
That's great for me! 

Should I close this issue?

ipa-mdl, 2020-05-28:
> Should I close this issue?

It would be great if you could rerun the broken test job first to confirm that it is fixed now after #517 was merged :)

ShotaAk, 2020-05-28:
I just re-ran the broken job and that results another error caused by missing a dependency of our package.
( This is our repository's problem)

I fixed only package.xml and the job succeeded.
#517 also worked! :+1: 

https://github.com/rt-net/raspimouse2/pull/20

closed 2020-05-28 by ipa-mdl