ros-industrial/industrial_ci #515: Official ROS docker images do not ship compilers anymore

ipa-mdl, 2020-05-28:
This breaks most ROS=repo builds

gavanderhoorn, 2020-05-28:
According to @ruffsl in [ROS Noetic Ninjemys Release](https://discourse.ros.org/t/ros-noetic-ninjemys-release/14262/3):

> Building tools (`rosdep`, `compilers`, `git`, `rosinstall` etc) are now in `ros:noetic-ros-base` and not part of the `ros-core` image anymore

ipa-mdl, 2020-05-28:
@gavanderhoorn: Thanks for pointing this out! Mostly just read new posts..

It looks like this affected all other images as well.

gavanderhoorn, 2020-05-28:
[[RFC] Restricting the size of ROS docker images](https://discourse.ros.org/t/rfc-restricting-the-size-of-ros-docker-images/13252) has some more details.

It wasn't communicated too well though imo.

ipa-mdl, 2020-05-28:
The `*-base` includes too many other dependencies, some undeclared dependencies would be shadowed..

My idea is now to install all needed tools "on-demand".
This way, `industrial_ci`  could be used with any Debian-based image and support use cases like #459

It is not much left anyway: https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/docker.sh#L235-L248

ruffsl, 2020-05-28:
> [RFC] Restricting the size of ROS docker images has some more details

pinging @mikaelarguedas , as may be of interest

mikaelarguedas, 2020-05-28:
Thanks for the ping,

Would a set of images similar to what is described at https://github.com/osrf/docker_images/issues/408 be of interest?

ipa-mdl, 2020-05-28:
@mikaelarguedas: Yes, that could be a reasonable default choice, at least for the more recent distros :)

mikaelarguedas, 2020-05-28:
Good to know.

> at least for the more recent distros :)

My initial feeling was to provide these images only for currently active ROS Distro/ OS Distro combinations but a case could be made for a wider set of tags.

If you have feedback or a wishlist of what tags / packages / configuration options you'd like to see in these images, please comment directly on the ticket

ipa-mdl, 2020-05-28:
> My initial feeling was to provide these images only for currently active ROS Distro/ OS Distro

Yes, that's fine :)

gavanderhoorn, 2020-05-29:
So, is [this](https://travis-ci.com/github/ros-industrial/abb_libegm/jobs/341441230) also related to this issue? Or is it something else?

ipa-mdl, 2020-05-31:
@gavanderhoorn:

It is a regression of #516 
And perhaps a problem with catkin-tools, which needs requires catkin for building non-catkin packages.

ipa-mdl, 2020-05-31:
> It is a regression of #516

It would have failed anyway without ROS_REPO=ros
#518 should fix it

gavanderhoorn, 2020-05-31:
> It would have failed anyway without ROS_REPO=ros

I'm not sure I understand what you mean by this: the repository which had a failed build does have `ROS_REPO=ros` in the `.travis.yml`?

ipa-mdl, 2020-06-08:
@gavanderhoorn:
Your repository has `ROS_REPO=ros` configured, which selects the official Docker image.
If you would have used the default (`ROS_REPO=testing`), then it would have built the custom Docker image and it would have failed even without the latest changes.

#525 works properly for both cases :)

PR #525 (2020-06-07): always install build-essential
reverts #516 
fixes #515
closed 2020-06-08 by ipa-mdl