ros-industrial/industrial_ci #519: Sourcing an environment setup script in a BEFORE_* script.

tylerjw, 2020-06-05:
I'm depending on a specific version of a library I have to manually install and as a part of that, I need to source a script that comes as part of that library.

The problem is that if I put this command in a `BEFORE_*` variable the symbols inside it do not make it into the shell that does the build.

Is there a way to achieve what I need that I'm not seeing?

ipa-mdl, 2020-06-07:
> Is there a way to achieve what I need that I'm not seeing?

No, the isolation is enforced.

> I need to source a script that comes as part of that library.

It should be possible to add handling for a special keyword like `sourcefile`, e.g.
```
AFTER_INIT="[install framework..]"
AFTER_AFTER_INIT="sourcefile /opt/framework/sourceme.bash param1 param2"
```
It would source the file into the current build environment.

tylerjw, 2020-06-07:
Fixed by https://github.com/ros-industrial/industrial_ci/pull/524

closed 2020-06-07 by tylerjw