ros-industrial/industrial_ci #523: rerun_ci doesn't copy .ssh directory into docker

tylerjw, 2020-06-07:
I need the copying of the ssh directory to get a key I need to checkout private repos inside the docker container.  run_ci does this but rerun_ci doesn't.

tylerjw, 2020-06-07:
This doesn't matter because it seems that ssh-agent socket forwarding is working in rerun-ci so I can do this to verify that it is working:

`AFTER_INIT='echo $SSH_AUTH_SOCK && ssh-add -l'`

In run_ci I had to do this to get it to work:

`ADDITIONAL_DEBS="openssh-client" AFTER_INIT='eval $(ssh-agent -s) && ls ~/.ssh && ssh-add ~/.ssh/id_rsa'`

Maybe something is broken in `run_ci`?  If not, close this.

ipa-mdl, 2020-12-13:
`HOME` was not passed as an environment, varibale so the keys folder could not be found..

closed 2020-12-14 by ipa-mdl

ipa-mdl, 2020-12-14:
To prevent the leak of SSH keys bunding the credentials will by an opt-in (#570)

PR #568 (2020-12-14): pass HOME from rerun_ci to run_ci
fixes #523