ros-industrial/industrial_ci #536: ccache directory not mounted in Gitlab docker:dind

Briancbn, 2020-06-21:
Here is my .travis.yml
```yml
image: docker:git
services:
  - docker:dind
before_script:
  - apk add --update bash coreutils tar curl ccache
  - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
  - mkdir -p $HOME/.ccache/
cache:
  key: ${CI_COMMIT_REF_SLUG}
  paths: 
    - $HOME/.ccache/

variables:
  CCACHE_DIR: "$HOME/.ccache"
  DOCKER_DRIVER: overlay2

dashing-main:
  script: .industrial_ci/gitlab.sh ROS_DISTRO=dashing ROS_REPO=main AFTER_SCRIPT="ccache -s"

dashing-testing:
  script: .industrial_ci/gitlab.sh ROS_DISTRO=dashing ROS_REPO=testing

eloquent-main:
  script: .industrial_ci/gitlab.sh ROS_DISTRO=eloquent ROS_REPO=main

eloquent-testing:
  script: .industrial_ci/gitlab.sh ROS_DISTRO=eloquent ROS_REPO=testing

after_script:
  - ls $HOME/.ccache
  - ccache -s

```
And here is the [log](https://gitlab.com/Brian97cbn/packml_ros2-gitlab/-/jobs/604649406). 
I have an after script to check the `$HOME/.ccache` folder, but it was empty.
The ccache directory is not mounted onto the path **CCACHE_DIR** specified. 
And rerun the CI pipeline won't use the correct cache to download

Briancbn, 2020-06-21:
According to this ticket https://gitlab.com/gitlab-org/gitlab-foss/-/issues/41227

`docker:dind` only allow the certain share directory to mount the cache folder. 
So the **CCACHE_DIR** should only set to `/builds/$CI_PROJECT_PATH/shared` in Gitlab CI when using `docker:dind`

ipa-mdl, 2020-06-21:
> docker:dind only allow the certain share directory to mount the cache folder.

That is not entirely true, AFAIK it can be configured for each runner..

> So the CCACHE_DIR should only set to /builds/$CI_PROJECT_PATH/shared in Gitlab CI when using docker:dind

Actually, any path under `/build` should work..

> It took a pretty long time to realize it today and I think it's worth documenting inside the documentation

I had the same problem with TMPDIR, so I added a comment in our sample config:
https://github.com/ros-industrial/industrial_ci/blob/master_transitional/.gitlab-ci.yml#L7

ipa-mdl, 2020-06-21:
There seems to be an additional restriction: https://docs.gitlab.com/ee/ci/yaml/README.html#cachepaths

Briancbn, 2020-06-22:
> Paths are relative to the project directory ($CI_PROJECT_DIR) and can’t directly link outside it.

Yes, thanks. it seems that `$CI_PROJECT_DIR` is actually `/builds/$CI_PROJECT_PATH/`. I will update the documentation PR accordingly

closed 2020-07-06 by ipa-mdl