ros-industrial/industrial_ci #542: Travis CI Foxy: RCC (the QT resource compiler) couldn’t find files

Briancbn, 2020-07-05:
Here is my compilation error log
```
utoRcc subprocess error

------------------------

The rcc process failed to compile

  "SRC:/resources/resource.qrc"

into

  "BIN:/traffic-editor_autogen/3YJK5W5UP7/qrc_resource.cpp"

Command

-------

/usr/lib/qt5/bin/rcc -name resource -o /root/upstream_ws/build/traffic_editor/traffic-editor_autogen/3YJK5W5UP7/qrc_resource.cpp /root/upstream_ws/src/traffic_editor/traffic_editor/resources/resource.qrc

Output

------

RCC: Error in '/root/upstream_ws/src/traffic_editor/traffic_editor/resources/resource.qrc': Cannot find file 'icons/vertex.svg'
```

It seems that qt5 rcc fails in both Foxy main and testing and it is happening to multiple packages of mine.
I tried the Foxy docker image locally, there isn't a compilation problem. 
Same problem doesn't occur in Gitlab CI either. I suspect it is due to something related to the docker image used by Travis.

Briancbn, 2020-07-05:
I managed to resolve this problem by setting `DOCKER_RUN_OPTS=--privileged` ([references](https://www.digitalrooster.dev/news/travis-updated-docker/)). I am not sure whether this should be added as the default run options, because Foxy is going to be more mainstream in the future.

gavanderhoorn, 2020-07-05:
> I managed to resolve this problem by setting `DOCKER_RUN_OPTS=--privileged` ([references](https://www.digitalrooster.dev/news/travis-updated-docker/)). I am not sure whether this should be added as the default run options, because Foxy is going to be more mainstream in the future.

No, not it should not.

Briancbn, 2020-07-05:
OK, thanks then I am closing this.

gavanderhoorn, 2020-07-05:
Did you solve your problem without `--privileged`?

If not, you should keep this open.

I just commented that adding `--priviliged` by default seems like something we should avoid *and* it seemed strange to me that a simple build step would require a `--privileged` docker container.

That's typically only needed to access resources *outside* of the container (ie: device files, memory segments, etc).

Briancbn, 2020-07-05:
> Did you solve your problem without `--privileged`?

No, only with privileged. OK, I am reopening this.

> That's typically only needed to access resources _outside_ of the container (ie: device files, memory segments, etc).

Thanks. I will look into this and open a PR. It seems that I need to whitelist statx syscall https://github.com/docker/for-linux/issues/208

gavanderhoorn, 2020-07-05:
>  It seems that I need to whitelist statx syscall [docker/for-linux#208](https://github.com/docker/for-linux/issues/208)

That's a nice find.

`statx` is indeed something which would need special privileges.

A more fine-grained approach than adding `--privileged` would be preferred, but if there is none, adding it using `DOCKER_RUN_OPTS` could make sense.

Setting it by default would not seem necessary here.

PR #543 (2020-07-06): Update libseccomp2 to fix statx syscall error

Fix #542 
This problem has been bugging me for some time since the foxy-nightly docker
This fix will update libseccomp2 to higher than 2.3.3.
This will enable the docker-ce 18.06.0 run in Travis CI to allow statx syscall, which will fix the Qt5 RCC not found error in Foxy.
Reference: https://travis-ci.community/t/docker-build-environments/7216/4
closed 2020-07-06 by ipa-mdl