ros-industrial/industrial_ci #550: repos in re-used docker image not being pulled

tylerjw, 2020-07-28:
When I do a rerunci job I'm regularly seeing this message:

```
Your branch is behind 'origin/develop' by 6 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
```

When I open the docker image for some reason those repos are pulling to be the latest.  I'll dig into this and try to fix this.

ipa-mdl, 2020-07-28:
This is an output of `git`. I don't see any issue here..
According to `https://github.com/dirk-thomas/vcstool/issues/165` you are using a commit as `version`, so this output is to be expected (?).

tylerjw, 2020-07-28:
The issue is that when I re-use a docker container for a CI run the version in my upstream repos never gets updated, even when they are old.  So if I depend on moveit and someone lands a PR on master when I rerunci it doesn't pull that commit, instead, it just continues on from wherever it had cached a version (in the docker container).

This is different from running a ci run from scratch because from scratch (or in the ci system) it will get the latest code.

ipa-mdl, 2020-07-28:
In your particular use case, I would just comment out the version locally..
This does not change the hash of `rerun_ci`.

Or you could add the new version after your repos file (changes the hash).

tylerjw, 2020-07-28:
Comment out what version?

This is a two-part problem.  The first is that the repos, when used with normal branches as versions are used, don't get updated to the latest when you rerunci.

The second is that this is not as simple as calling `vcs pull` after every call of `vcs import`.  That is why I used a custom command that calls checkout (does the same thing, it just doesn't expect HEAD to be tied to a branch like pull does).

ipa-mdl, 2020-07-30:
I am still trying to understand the issue (and where the root cause is located).

I would expect the following:
- if `version` is a commit or tag, a subsequent call to `vcs import --force` should not change anything
- if `version` is a branch, a subsequent call to `vcs import --force` should update to the latest remote branch

Given your comments and the simple test I ran,latter does not seem to be the case..
Not sure if this a bug in vcstool and by design.
@dirk-thomas: what is the expected behavior for using `vcs import --force` with the same repos file, but updated remote branches?

dirk-thomas, 2020-07-30:
A repeated `vcs import` won't necessarily update an existing working copy to the latest remote state. In case the working is on the same branch already it will only `fetch` from the remote. Atm a subsequent `vcs pull` is needed for that.

`--force` doesn't have any effect if the existing repo URL matches the entry in the repos file.

ipa-mdl, 2020-07-30:
@dirk-thomas: Thanks for the clarification! Unfortunately, `vcs pull` fails (stderr messages and exit code), if the the workspace contains commit-based working copies (That's what https://github.com/dirk-thomas/vcstool/issues/165 is about)

dirk-thomas, 2020-07-30:
> `vcs pull` fails (stderr messages and exit code), if the the workspace contains commit-based working copies. (That's what dirk-thomas/vcstool#165 is about)

Please follow the conversation on that ticket which also references a pull request to address exactly that.

closed 2021-07-21 by ipa-mdl