ros-industrial/industrial_ci #553: [gitlab] access to private repos in upstream workspace

fmessmer, 2020-08-19:
I'm trying to access private gitlab repositories while setting up the upstream workspace:
I'm not super-experienced with `gitlab` and I don't think all the steps from https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#gitlab-ci-access-to-private-repositories have been performed yet

But looking at the log output:
```
=== /root/upstream_ws/src/XXX (git) ===
  Could not determine ref type of version: error: cannot run ssh: No such file or directory
  fatal: unable to fork
```
it seems that `ssh` is not even available at the time

the `.gitlab-ci.yml` looks like this:
```
variables:
  # needed for docker-in-docker
  IMAGE_TAG:          $CI_REGISTRY_IMAGE/$CI_COMMIT_REF_SLUG
  DOCKER_TLS_CERTDIR: ""
  
  # industrial_ci global variables
  TMPDIR: "${CI_PROJECT_DIR}.tmp"
  CCACHE_DIR: ${CI_PROJECT_DIR}/ccache
  
  # industrial_ci specific test setup
  CATKIN_LINT: pedantic
  CATKIN_LINT_ARGS: '--ignore description_meaningless --ignore description_boilerplate'
  CMAKE_ARGS: -DCMAKE_BUILD_TYPE=Release
  UPSTREAM_WORKSPACE: .gitlab-ci.rosinstall

image: docker:git

services:
  - docker:dind

before_script:
  # needed for docker-in-docker
  - docker login -u $CI_REGISTRY_USER -p $CI_BUILD_TOKEN $CI_REGISTRY

  # industrial_ci config
  - apk add --update bash coreutils tar grep # install industrial_ci dependencies
  - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master

kinetic:
  script: .industrial_ci/gitlab.sh ROS_DISTRO=kinetic

melodic:
  script: .industrial_ci/gitlab.sh ROS_DISTRO=melodic
```

and the `UPSTREAM_WORKSPACE` is specified via a rosinstall file

fmessmer, 2020-08-20:
from the logs, I assume that the `ici_install_pkgs_for_command ssh ssh-client` step is skipped:
https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/workspace.sh#L90-L92
```
function ici_setup_git_client {
  ici_install_pkgs_for_command git git-core
  if [ -d ~/.ssh ]; then
    ici_install_pkgs_for_command ssh ssh-client
  fi
}
```
thus, `ssh` actually not being installed

fmessmer, 2020-08-20:
apparently, all variables were configured correctly on the master repo - but when running CI on a PR coming from a branch of a fork of the master repo, the CI runner runs on the forks settings - and there the variables are not set/have to be set manually

is there a way to "inherit" the keys `SSH_PRIVATE_KEY` and `SSH_SERVER_HOSTKEYS` to the forks?

Briancbn, 2020-08-21:
Hi @fmessmer, not sure if this help, you can configure your gitlab_ci to run **detached merge request**, which can still be run by the master repo CI runner, but only if you are a developer. Here is the [detailed link](https://docs.gitlab.com/ee/ci/merge_request_pipelines/), I think you can simply test by adding the following lines to the start of your `.gitlab_ci.yml` file.
```yml
workflow:
  rules:
    - if: $CI_MERGE_REQUEST_ID
    - if: $CI_COMMIT_TAG
    - if: $CI_COMMIT_BRANCH
```

fmessmer, 2020-09-01:
issue was solved for PRs from master repo by following the instructions in the README
access from forks will be solved by setting up the gitlab runner accordingly

closed 2020-09-01 by fmessmer