ros-industrial/industrial_ci #562: UPSTREAM WORKSPACE syntax for multiple repos?

tianshiz, 2020-10-01:
Hi, I'm struggling a bit to specify two separate source repos as my upstream.
I use bitbucket pipelines.
Originally I have UPSTREAM_WORKSPACE defined as:
`github:stereolabs/zed-ros-wrapper#v3.0.3`

Now I want to add another upstream, this time from a bitbucket repo that is private but under the same account as the pipelines. How do I add this repo? I tried:
`github:stereolabs/zed-ros-wrapper#v3.0.3 bitbucket:myaccount/myrepo`

But I'm getting errors that it cannot parse and:
`Input data is not valid yaml format: unacceptable character #x001b: special characters are not allowed`

Whats the way to do this?

ipa-mdl, 2020-10-01:
> How do I add this repo? I tried:
> `github:stereolabs/zed-ros-wrapper#v3.0.3 bitbucket:myaccount/myrepo`

Exactly like this.. 

> But I'm getting errors that it cannot parse and:
Input data is not valid yaml format: unacceptable character #x001b: special characters are not allowed

I can confirm this error, I will have a look.

ipa-mdl, 2020-10-01:
Please try `UPSTREAM_WORKSPACE="github:stereolabs/zed-ros-wrapper#v3.0.3 bitbucket:myaccount/myrepo#HEAD"`.
As [documented](https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#workspace-definition), the version fragment is mandatory.

However, the error strange is not very helpful and I am not sure why it is printed at all.

ipa-mdl, 2020-10-01:
With #563, you will get the following error message:
> Could not parse URL 'bitbucket:myaccount/myrepo'. It does not match the expected pattern: \<scheme>:\<resource>#\<version>.
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'setup_upstream_workspace' returned with code '1' after 0 min 18 sec

closed 2020-12-13 by ipa-mdl