ros-industrial/industrial_ci #565: Docker Pull Rate Limit Enforcement from November 2, 2020, 17:00 UTC

ipa-mdl, 2020-10-28:
Docker Service Announcement, October 28, 2020:

> On Monday, November 2, 2020 at 9am Pacific Standard Time, Docker will begin enforcing rate limits on container pulls for Anonymous and Free users. Anonymous (unauthenticated) users will be limited to 100 container image pulls every six hours, and Free (authenticated) users will be limited to 200 container image pulls every six hours, when enforcement is fully implemented.

This restriction will be based on IPs, so it will most likely affect all publicly shared runners/worker on most CI/CD services.

tianshiz, 2020-11-05:
I'm getting this:
"E: Failed to fetch https://packagecloud.io/github/git-lfs/ubuntu/dists/bionic/InRelease  402  Payment Required [IP: 54.183.38.243 443]
E: The repository 'https://packagecloud.io/github/git-lfs/ubuntu bionic InRelease' is no longer signed."

is this related? Haven't run my pipelines since last week

ipa-mdl, 2020-11-05:
> is this related?

Probably not. The rate limiting only applies to `docker pull`.

ipa-mdl, 2020-12-13:
It looks like the major CI services have prepared counter measures.

For local runners a [pull-through registry cache](https://docs.docker.com/registry/recipes/mirror/) should do the trick.

closed 2020-12-13 by ipa-mdl