ros-industrial/industrial_ci #567: [github ci] "/bin/bash: /root/ici/ci_main.sh: No such file or directory" error with self-hosted runner

eliabruni, 2020-11-29:
Hi all,

This is great work and would like to use it in our ci pipeline. I saw that the same error was reported in #213 for gitlab, however I cannot figure out how to adapt the solution to my case.

 We have GitHub [self-hosted](https://docs.github.com/en/free-pro-team@latest/actions/hosting-your-own-runners) arm64 runners. I therefore tried to use your action, with the following `.yml` file:

```
# This config uses industrial_ci (https://github.com/ros-industrial/industrial_ci.git).
# For troubleshooting, see readme (https://github.com/ros-industrial/industrial_ci/blob/master/README.rst)

name: CI

# This determines when this workflow is run
on: [push, pull_request] # on all pushes and PRs

# or more fine-grained
# on:
#   # When master or melodic-devel branch is pushed to
#   push:
#     branches: [ master, melodic-devel ]
#   # When there is a pull request against master
#   pull_request:
#     branches: [ master ]

jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - {ROS_DISTRO: melodic, ROS_REPO: testing}
          - {ROS_DISTRO: melodic, ROS_REPO: main}
    env:
      CCACHE_DIR: /github/home/.ccache # Directory for ccache (and how we enable ccache in industrial_ci)
    runs-on: [self-hosted, linux, ARM64]
    container:
      image: ***/cirunner
      options: -v /var/run/docker.sock:/var/run/docker.sock
      
    steps:
      - uses: actions/checkout@v2
      # This step will fetch/store the directory used by ccache before/after the ci run
      - uses: actions/cache@v2
        with:
          path: ${{ env.CCACHE_DIR }}
          key: ccache-${{ matrix.env.ROS_DISTRO }}-${{ matrix.env.ROS_REPO }}
      # Run industrial_ci
      - uses: 'ros-industrial/industrial_ci@master'
        env: ${{ matrix.env }}
```

Unfortunately, I am getting the following error:

```
Run ros-industrial/industrial_ci@master
/usr/bin/docker exec  06dff26795df4e48502bc723ddcca3b09b559126bd793cedd214f697f90b9201 sh -c "cat /etc/*release | grep ^ID"

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function 'prepare_docker_image' returned with code '0' after 2 min 49 sec
/bin/bash: /root/ici/ci_main.sh: No such file or directory
```
 
Any help would greatly appreciated!

ipa-mdl, 2020-12-13:
`/root/ici/` gets bind-mounted from the clone GitHub action. It might be that your local runner (or the container) need an additional volume to make that work.

alvarozornoza, 2021-01-04:
I have same problem that @eliabruni has.

Shall you extend your answer @ipa-mdl ? Do you know how to configure the docker volume you mention in your answer? Further explanation would be kindly appreciated :100: 

Thanks in advance

ipa-mdl, 2021-01-04:
>  Do you know how to configure the docker volume you mention in your answer? Further explanation would be kindly appreciated 

Unfortunately, I do not know how to configure it. I am not using self-hosted Github runners.

eliabruni, 2021-01-05:
@alvarozornoza check [here](https://github.com/ros-industrial/industrial_ci/issues/574#issuecomment-748841549)

closed 2021-01-05 by eliabruni