ros-industrial/industrial_ci #575: Mixed ROS Versions

dave992, 2020-12-15:
A question I just asked in the ROS-Industrial Conference talk about Software Quality:

> How should dependencies be handled which use other ROS versions? E.g. my package runs on Melodic, but depends on a package that runs only on Noetic and another which runs only on Kinetic. To run this I use docker images

I have tried to implement industrial_ci for packages that serve as an 'execution manager' and as a result depend on a lot of separate packages.  Some of these packages run a lot easier on different ROS versions as their dependencies are already part of the respective Ubuntu/ROS version. Examples are `ros_industrial/noether` or packages reliant on `OpenCV4`, which are more easily used on ROS Noetic. 

However, it is not always viable to immediately switch over to the new Ubuntu/ROS version as it is released and some packages just do not work on newer ROS versions. (This might be less of a problem now than a few months ago.) 

My approach now has been to build a custom Docker image for the CI with all the dependencies manually installed and building some packages from source. For 1 (or 2) packages this is doable, but it quickly gets an annoyance and in some cases, I have not been able to get the Docker image to work correctly. This also is not how the application is used normally, where we just spin up a Docker image for those packages. 

Is there a more convenient way to implement `industrial_ci` for applications where packages depend on multiple ROS Versions?

ipa-mdl, 2020-12-15:
`industrial_ci` is mostly meant for testing the same code for different ROS version, but not for mixing them.

It is possible to create Docker images for CI, but AFAIK there is no simple way to test such packages.

If you already have your images, you can just spin them up in CI.
Your actual test should then be run directly by [reusing the running ROS master](http://wiki.ros.org/rostest/Connecting%20to%20the%20rostest%20master): `rostest --reuse-master foo_package foo.test`.

So roughly:
* ./ci.sh TARGET_WORKSPACE="..." ROS_DISTRO=melodic DOCKER_COMMIT=test-melodic
* ./ci.sh TARGET_WORKSPACE="..." ROS_DISTRO=noetic DOCKER_COMMIT=test-noetic
* docker run -d --rm test-melodic ... roslaunch ...
* docker run -d --rm test-noetic ... roslaunch ...
* ./ci.sh ROS_DISTRO=noetic AFTER_SCRIPT="rostest --reuse-master foo_package foo.test"
* docker stop -f  test-melodic test-noetic

To be honest it is not clear to me why someone would need to run a system test like this.
It might be much simpler to just use a rosbag or a stub.

dave992, 2020-12-16:
I may have not explained it well enough :). I still want to test the same code for different ROS versions, but my question is just regarding the dependencies which are not usable (or harder to use) on different ROS versions. 

I use `industrial_ci` for package A on GitHub/GitLab, which has external dependencies on package B, C, and D which need to be build from source. 
- Package B runs only on ROS Noetic
- Package C runs only on ROS Melodic
- Package D runs on multiple versions

To run this application locally:
- I run package A on the system ROS version (melodic) and spin up a docker image for package B from which I launch package B.
- I run package A on the system ROS version (noetic) and spin up a docker image for package C from which I launch package C. 

Ideally, I would port all packages to a single version and make sure the local systems use the same version. Unfortunately, this is not always feasible hence my question if there is a more straightforward way to test your package which has external dependencies that use a fixed and different ROS version than you are testing for.

gavanderhoorn, 2020-12-16:
@dave992: you're mostly interested in this for integration tests I guess?

You can of course not build binaries using artefacts from different Docker containers.

dave992, 2020-12-16:
@dave992: you're mostly interested in this for integration tests I guess?

Yes, indeed :)

ipa-mdl, 2020-12-16:
>  mostly interested in this for integration tests

And that's what `industrial_ci` does not deal with, but delegates to the ROS test system.
`industrial_ci` is mostly concerned with building the packages in the right environment.

> hence my question if there is a more straightforward way to test your package which has external dependencies that use a fixed and different ROS version than you are testing for.

Did you consider wrapping the docker image in a dummy package for the unsupported distros?
This is how I made `noether` work in a past project..

ipa-mdl, 2020-12-16:
> This is how I made noether work in a past project..

```shell
#!/bin/bash
exec docker run --rm --net=host -e ROS_NAMESPACE -e ROS_MASTER_URI <YOUR_IMAGE> rosrun <YOUR_PACKAGE> <YOUR_NODE> "$@"
```

gavanderhoorn, 2020-12-16:
With some more work you can even do this using a `launch-prefix`.

dave992, 2020-12-16:
I see that I mixed some things up when I went from 'building everything in a single workspace' to 'separating the troublesome packages in docker images' and how I should use `industrial_ci` in both cases. Nevertheless still very helpful, so thanks for the feedback!

closed 2020-12-16 by dave992

ipa-mdl, 2020-12-16:
> With some more work you can even do this using a launch-prefix.

Does roslaunch check if the package and especially the node exist?

gavanderhoorn, 2020-12-16:
I don't remember running into difficulties with that, but I could be misremembering things.