ros-industrial/industrial_ci #578: Collect JUnit XML test results for GitLab

stertingen, 2021-01-11:
GitLab has a feature to collect JUnit XML test results and display them in the Pipeline UI: https://docs.gitlab.com/ee/ci/unit_test_reports.html

Currently, this is not directly supported/documented by the CI, but the following `.gitlab-ci.yml` does the job:

```yaml
image: docker:git

variables:
  DOCKER_RUN_OPTS: "-v ${CI_PROJECT_DIR}/.build:/root/target_ws/build" # Share build space

services:
  - docker:19.03.5-dind

before_script:
  - apk add --update bash coreutils tar grep
  - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master

# Collect test results by default
default:
  artifacts:
    when: always
    reports:
      junit: ${CI_PROJECT_DIR}/.build/**/test_results/**/*.xml

kinetic:
  script: .industrial_ci/gitlab.sh
  variables:
    ROS_DISTRO: "kinetic"
```

It places the build space for the target workspaces in the outer docker container and mounts it into the inner container. The extra mount seems a little bit hacky to me, but it works fine.

I thought that others might profit from this, so this could be added to the README or the example config.

ipa-mdl, 2021-01-11:
Very nice!

>  The extra mount seems a little bit hacky to me

It is not too hacky.. just one of the downside of using Docker (or containers in general).
With https://github.com/ros-industrial/industrial_ci/pull/572, you can even avoid the mount :)

> I thought that others might profit from this, so this could be added to the README or the example config.

Definitely! I will add it to https://gitlab.com/ipa-mdl/industrial_ci
Anyway, the example config is a little bit outdated now and needs some overhaul (like most of the documentation..).

ipa-mdl, 2021-01-12:
I have now added support for mounting the workspaces automatically (https://github.com/ros-industrial/industrial_ci/pull/572).
```yaml
test_junit:
  script: .industrial_ci/gitlab.sh
  variables:
    ROS_DISTRO: noetic
    BASEDIR: ${CI_PROJECT_DIR}/.build
  artifacts:
    when: always
    reports:
      junit: ${BASEDIR}/**/test_results/**/*.xml
```

(results: https://gitlab.com/ipa-mdl/industrial_ci/-/pipelines/240764976/test_report)

stertingen, 2021-01-13:
> I have now added support for mounting the workspaces automatically (#572).
>
> ```yaml
> test_junit:
>   script: .industrial_ci/gitlab.sh
>   variables:
>     ROS_DISTRO: noetic
>     BASEDIR: [...]

Very nice! Looking forward to see this in `master`.

Currently your changes are available on GitLab only. Is that intended?

(P.S.: I find `.build` as name for the base directory rather distracting since it does not refer to a single build space.)

ipa-mdl, 2021-01-13:
> Currently your changes are available on GitLab only. Is that intended?

`BASEDIR` will be available everywhere..
Other CI services might not have the same mount issue, though.

>(P.S.: I find .build as name for the base directory rather distracting since it does not refer to a single build space.)

".build" is meant in the sense of CI build
Any better idea? ;)

stertingen, 2021-01-13:
> > (P.S.: I find .build as name for the base directory rather distracting since it does not refer to a single build space.)
> 
> ".build" is meant in the sense of CI build
> Any better idea? ;)

`workspaces`, so the paths would be `$HOME/workspaces/target_ws/`, `$HOME/workspaces/upstream_ws` and so on.

ipa-mdl, 2021-01-13:
@stertingen: It turned out that I forgot to push it to Github ;)
The feature is now contained in #572 and is using `BASEDIR: ${CI_PROJECT_DIR}/.workspaces`.

ipa-mdl, 2021-01-16:
`BASEDIR` support is not merged, I will keep this issue open, because it still needs to be documented.

stertingen, 2021-01-26:
`BASEDIR: ${CI_PROJECT_DIR}/...` does not work when the target workspace is in the repository, because then the CI script tries to copy the `${CI_PROJECT_DIR}` into `${BASEDIR}/target_ws/src`, i.e. into itself. Unfortunately, artifacts have to be stored inside of `${CI_PROJECT_DIR}`, so setting `BASEDIR` to something like `/basedir` does also not work.

ipa-mdl, 2021-01-27:
> BASEDIR: ${CI_PROJECT_DIR}/... does not work when the target workspace is in the repository

I did not consider this in the first place..
Please test #595

stertingen, 2021-01-27:
> 
> 
> > BASEDIR: ${CI_PROJECT_DIR}/... does not work when the target workspace is in the repository
> 
> I did not consider this in the first place..
> Please test #595

Works like a charm, thank you!

stertingen, 2021-02-17:
I don't know if you read this in [the other issue](https://github.com/ros-industrial/industrial_ci/pull/595#issuecomment-779171664): Setting `BASEDIR` currently only works for `ISOLATION=docker`, unfortunately not for `ISOLATION=shell`. It's the same issue: The CI script tries to copy a directory into itself.